alibaba/AliOS-Things #3: 确实编译文档

spyhole, 2017-10-21:


oujunli, 2017-10-21:
参考 [AliOS Things Environment Setup](https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-Environment-Setup)

Haleyl, 2017-10-21:
资料确实有点少

lluwang, 2017-10-21:
@spyhole 请问[https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-Environment-Setup](url)是否满足你的编译文档需求？
@Haleyl 谢谢你的意见，请问是否烦请给出具体的意见，觉得哪块文档缺失？谢谢

spyhole, 2017-10-22:
跟随你建议先

Haleyl, 2017-10-22:
1，关于内核的讲解和线程的调度方面，如果有框架图会更好理解系统；
2，看资料都是以API展开，实例很少，如果是按照VFS组件的统一接口说法，其实我们并不关心HAL层；
3，支持的bsp还是很少；
最后一点，目录结构看着有点纠结，组件和bsp剥离不够彻底啊

tidyjiang8, 2017-10-23:
文档这块，确实挺少的，我觉得zephyr的文档就做得特别好
- https://www.zephyrproject.org/doc/index.html 
- http://zephyr-doc.readthedocs.io/zh_CN/latest/index.html

dongAxis, 2017-10-23:
嵌入MK3060的硬件外壳叫啥名字？

dongAxis, 2017-10-23:
硬件环境搭建的外设板能直接买到？

tidyjiang8, 2017-10-23:
@dongAxis 看我提的另一个issue https://github.com/alibaba/AliOS-Things/issues/8

woksdl, 2017-10-23:
@Haleyl 组件和bsp剥离不够彻底 这部分具体指什么啊

Haleyl, 2017-10-23:
@woksdl 比如stm32l4xx/bsp目录下的Components，这些组件本是和stm32l4xx无关的，可以剥离出来让所有bsp共用。

kejinlu, 2017-10-26:
@lluwang  仔细看这些文档，你会发现 上下文的那些截图很多都匹配不上，你看那些截图的里面的端口号，上下文都对不上，一会儿设备管理器里面是COM3 COM5，然后一会儿连接串口的终端配置的又是COM7

singerswong, 2019-01-21:
关了这issue吧，我现在对阿里是无比的伤心，不是因为这一个 AliOS-Things 项目，大家别误会，使整个阿里价值观

closed 2019-01-21 by singerswong