alibaba/AliOS-Things #7: 物联网任重道远啊

liudf0716, 2017-10-23:
大致看完readme，对alios-things具体干啥的一头雾水。建议给个实际的案例说明，而不是仅仅宽泛的说明我们啥都能干。
另外alios-things跟openwrt，lede比有啥优势呢？还是说大家瞄准的方向不一样。

wbangin, 2017-10-23:
差别比较大，openwrt，是基于linux，一般运行于SOC芯片之上的系统，资源占用比较大。alios-thigns是运行于mcu上面，内存占用只有几十K的rtos。

gengzhaoyun, 2017-10-23:
这个跟ucos相比，区别在哪里呢？

sonwa, 2017-10-23:
有没有实际例子？

tidyjiang8, 2017-10-23:
@gengzhaoyun ucos纯粹是个rtos，是个kernel，而我的理解，alios-things与ucos的一个重要区别是它包含一个可以**直接连接阿里云的协议栈**，此外：
- alios-things 跨平台，支持winows+linux，而且是两个平台都支持得比较好，这一点甚是让人喜欢，相比较其它iot rots，例如liteos仅对windows支持得比较好，zephyr仅对linux支持得比较好
- 增加了各种辅助功能

tidyjiang8, 2017-10-23:
@sonwa 看example目录啊

woksdl, 2017-10-23:
fota服务（多bin/压缩），掉电保护的文件系统、trustzone上的teeos等物联网需要的必备组件

closed 2022-03-01 by skylarCai

sonwa, 2022-03-01:
您好，我已经收到你刚刚发来的邮件，我会尽快回复！
The e-mail you sent to me has been received,I will check it as soon as possible,thank you.这一封邮件是系统自动回复。This e-mail is automatically reply.