alibaba/AliOS-Things #8: 如何购买开发板(mk3060模组)？

tidyjiang8, 2017-10-23:
前几天回老家了，一直没试，趁着上午，看了下文档，搭了下环境，跑了下linux下的helloworld，还没有深入阅读代码，感觉挺酷的，尤其是以VS Code作为IDE这块!

不过我想在真实的硬件上跑，但是没看到**在哪里购买mk3060模组呢**？

另外，这个OS今后会支持哪些板子？看到有个哥们再问是不是计划支持ESP系列，我也很期待...

tidyjiang8, 2017-10-23:
已找到购买链接，淘宝搜 _MiCOKit3060_

rockzhouchina, 2017-10-23:
https://item.taobao.com/item.htm?spm=a1z09.2.0.0.4db6f7af99FiSP&id=559865094527&_u=nm4sjv3cab
我刚购买

tidyjiang8, 2017-10-23:
@rockzhouchina 我也刚买了，thanks

yuguitao, 2017-10-24:
有跑起来的硬件组合图吗？  可以发来参考参考

tidyjiang8, 2017-10-24:
@yuguitao 我的板子估计今天上午到，计划到时候写一些博客 http://iot-fans.xyz :smiley:

kejinlu, 2017-10-25:
昨晚我试了下 死活烧不进去

lluwang, 2017-10-25:
@kejinlu，请问你的操作步骤是什么？

tidyjiang8, 2017-10-25:
@lluwang 
我在linux平台下能烧写成功，但是**上电后串口打印的是乱码**，各种常见的波特率都试过，没想明白哪儿有问题，这板子资料好少啊。
openocd的log文件内容：
```
work@ubuntu:~/me/AliOS-Things$ cat out/openocd.log 
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
adapter speed: 10000 kHz
none separate
use of EmbeddedICE dbgrq instead of breakpoint for target halt enabled
dcc downloads are enabled
force hard breakpoints
Info : No device selected, using first device.
Info : J-Link ARM V8 compiled Jan 31 2018 18:34:52
Info : Hardware version: 8.00
Info : VTarget = 3.313 V
Info : clock speed 10000 kHz
Info : JTAG tap: beken7231.cpu tap/device found: 0x15968001 (mfg: 0x000 (<invalid>), part: 0x5968, ver: 0x1)
Info : Embedded ICE version 6
Info : beken7231.cpu: hardware has 2 breakpoint/watchpoint units
requesting target halt and executing a soft reset
target halted in ARM state due to debug-request, current mode: Supervisor
cpsr: 0x000000d3 pc: 0x00000000
Error: No working memory available. Specify -work-area-phys to target.
Warn : not enough working area available(requested 24)
Info : no working area available, falling back to memory writes
304324 bytes written at address 0x00012000
downloaded 304324 bytes in 6.401293s (46.427 KiB/s)
	shutdown command invoked
```

tidyjiang8, 2017-10-25:
@kejinlu 
- **烧写时加参数 VERBOSE=1 看以看到详细的烧写命令**
- **out/openocd.log 文件中存放了烧写程序时的log**，你看里面是啥，如果提示类似 `error while loading shared libraries: libhidapi-hidraw.so.0`的话，需要先安装openocd:
**`$ sudo apt-get install openocd`**

oujunli, 2017-10-25:
可以做如下尝试：
1、如果是Windows，确认JTAG驱动版本，比较稳定版本是4.8，并且
2、进入bootloader模式下烧写试试。（进入bootloader的方式：按住boot键，按一下reset键）
@kejinlu

kejinlu, 2017-10-25:
我jlink还没到，所以直接 miniusb串口的 

按照文档点击upload 出现如下信息

`[INFO-Build] Run "C:\Users\Luke\.vscode\extensions\alios.alios-studio-0.4.2\bin\aos_64.exe make helloworld@mk3060 JTAG=jlink download" on c:\Users\Luke\Documents\Projects\AliOS-Things
[INFO-Build] -------------OUTPUT START-------------
[INFO-Build] Build AOS Now
[INFO-Build] 
[INFO-Build] Skipping building bootloader due to \"total\" is not set
[INFO-Build] 
[INFO-Build] 
[INFO-Build] 
[INFO-Build] 
[INFO-Build] AOS MEMORY MAP
[INFO-Build] |=================================================================|
[INFO-Build] | MODULE                                   | ROM       | RAM      |
[INFO-Build] |=================================================================|
[INFO-Build] | armv5                                    | 460       | 0        |
[INFO-Build] | beken                                    | 2254      | 4382     |
[INFO-Build] | board_mk3060                             | 511       | 16       |
[INFO-Build] | cli                                      | 6108      | 124      |
[INFO-Build] | digest_algorithm                         | 112       | 0        |
[INFO-Build] | framework                                | 376       | 12       |
[INFO-Build] | hal                                      | 172       | 12       |
[INFO-Build] | helloworld                               | 130       | 0        |
[INFO-Build] | kv                                       | 2977      | 36       |
[INFO-Build] | libalicrypto                             | 4078      | 0        |
[INFO-Build] | libbeken                                 | 163390    | 51833    |
[INFO-Build] | libc_nano                                | 18288     | 464      |
[INFO-Build] | libgcc                                   | 6088      | 0        |
[INFO-Build] | libmbedcrypto                            | 3582      | 8748     |
[INFO-Build] | libmbedtls                               | 12067     | 0        |
[INFO-Build] | libmesh                                  | 520       | 396      |
[INFO-Build] | log                                      | 40        | 4        |
[INFO-Build] | net                                      | 43434     | 3374     |
[INFO-Build] | newlib_stub                              | 240       | 0        |
[INFO-Build] | rhino                                    | 23673     | 6313     |
[INFO-Build] | vcall                                    | 6659      | 4        |
[INFO-Build] | vfs                                      | 1819      | 1109     |
[INFO-Build] | yloop                                    | 1311      | 24       |
[INFO-Build] | *fill*                                   | 335       | 103      |
[INFO-Build] |=================================================================|
[INFO-Build] | TOTAL (bytes)                            | 298624    | 76954    |
[INFO-Build] |=================================================================|
[INFO-Build] 
[INFO-Build] Downloading Bootloader ...
[INFO-Build] 
[INFO-Build] Skipping download bootloader due to \"total\" is not set
[INFO-Build] 
[INFO-Build] 
[INFO-Build] 
[INFO-Build] Downloading application to partition: 2 size: 304316 bytes...
[INFO-Build] 
[INFO-Build] Download failed
[INFO-Build] 
[INFO-Build] Build complete
[INFO-Build] 
[INFO-Build] Making .gdbinit
[INFO-Build] 
[INFO-Build] Task Complete!
[INFO-Build] -------------OUTPUT FINISHED-------------
[INFO-Device] Burn image c:\Users\Luke\Documents\Projects\AliOS-Things\out\helloworld@mk3060\binary\helloworld@mk3060.ota.bin on Device
[INFO-Device] Load Image File c:\Users\Luke\Documents\Projects\AliOS-Things\out\helloworld@mk3060\binary\helloworld@mk3060.ota.bin size 323340 bytes.
[INFO-Device] Timeout waiting device ask for transition.
[ERROR-Device] Fail to burn image c:\Users\Luke\Documents\Projects\AliOS-Things\out\helloworld@mk3060\binary\helloworld@mk3060.ota.bin due to Timeout waiting device ask for transition. Please make sure your board is in bootloader mode`

kejinlu, 2017-10-25:
mk3060的开发板按照你们文档以及官方文档尝试进bootloader模式（其实都不知道有没有进入），然后vscode点击那个upload按钮。出现上面的信息，看到连个问题 一个是 
`Skipping download bootloader due to "total" is not set ` 的信息，另一个就是 `Please make sure your board is in bootloader mode` ，都不知道是我操作不对，还是板子有问题，还是什么问题

kejinlu, 2017-10-25:
@oujunli  1. 我仿真器还没到，就先usb串口尝试的  2. https://item.taobao.com/item.htm?spm=a1z09.2.0.0.4db6f7af99FiSP&id=559865094527&_u=nm4sjv3cab 买的mk3060 只有boot按钮，没有所谓的reset按钮，alios things 里面标的reset按钮其实对应的是板子的chip_en 按钮。所以就纠结了，mk3060官方稳定说是按住boot按钮，通电 就是bootloader模式了。

kejinlu, 2017-10-25:
@tidyjiang8 我是在windows上 usb串口的方式进行的

kejinlu, 2017-10-28:
JLink可以download了，usb串口不行，希望官方给个说明。

junjiec, 2017-10-28:
@kejinlu 和开发板的bootloader有关，我同事给你的gmail邮箱发了邮件注意查收

kejinlu, 2017-10-29:
@junjiec 好的  感谢

TestDevelopment, 2017-12-15:
F:\saomiao\ALIOS\AliOS-Things-master>arm-none-eabi-gdb -x=.gdbinit -ex 'target r
emote localhost:3333' ./out/eclipse_debug/last_built.elf --tui
C:\Program Files (x86)\GNU Tools ARM Embedded\5.4 2016q3\bin\arm-none-eabi-gdb.e
xe: TUI mode is not supported

junjiec, 2017-12-19:
@TestDevelopment windows版本的gdb不支持tui

closed 2017-12-19 by junjiec