alibaba/AliOS-Things #23: 对文档的一些建议和改动

tidyjiang8, 2017-11-03:
AliOS的文档目前是托管在仓库的Wiki上面的，但是个人觉得查阅起来没那么方便，因此用 Shpinx 框架对文档做了些改动：
- 将其源码放到了 doc 目录下（ 点[这里](https://github.com/tidyjiang8/AliOS-Things/tree/doc-test/doc)查看）
- 将生成的页面托管在 [http://alios-things-doc-test.readthedocs.io](http://alios-things-doc-test.readthedocs.io) 上面了
- 将文档源码由 markdown 变为 reStructuredText 了（主要是为了支持 Sphinx 的更多功能）

**这样修改后的一些好处**：
- 可以通过doxygen自动从源代码中抽取注释并形成API文档，避免手动写ＡＰＩ文档，最终自动生成的ＡＰＩ文档的效果可以看 [这里](http://alios-things-doc-test.readthedocs.io/en/latest/api/kernel.html)
- 文档与源码维护在同一个仓库中，同时更新，防止今后出现文档、源码不匹配的现象
- 由于是在同一个仓库中，今后每次Release新版本时都会有对应该版本的文档
- 可以直接生成离线文档，例如 PDF、ＨＴＭＬ、ＥＰＵＢ等
- 如果今后有英文文档，也可以很方便地进行管理
- 文档页内可以自动生成目录，整体也可以自动生成目录，且可折叠，方便查看文档
- 可以对整个文档进行搜索

**不过这样也有一些不好的地方**，例如：
- 有少量ＡＰＩ的各式并非完全遵循Doxygen格式，因此需要做少量修改
- reStructuredText 的语法跟 markdown 的语法有些区别

目前只是把基本架子搭建好了： http://alios-things-doc-test.readthedocs.io
如果觉得这样做有意义的话，我就继续做
如果觉得这样做没啥意义的话，我就不继续做下去了

junjiec, 2017-11-03:
太牛了!我们内部评估一下

Cheng-SG, 2017-11-03:
太棒了，赞一个👍

baiyuxiong, 2017-11-04:
项目初期，这不是产生核心竞争力的地方。我觉得把精力花在补充新文档上，比做这个更有价值。

tidyjiang8, 2017-11-04:
> 项目初期，这不是产生核心竞争力的地方。我觉得把精力花在补充新文档上，比做这个更有价值。

@baiyuxiong 

YES，我也一直觉得文档比较少，只够用来入门，如果想要深入学习的话，只有去看代码了(但是如果有文档配合，看代码的效率会高很多)，所以应该扩展新文档，不过像我们这种非AliOS内部开发人员的话，写新文档感觉不现实，所以只能做点力所能及的事。

woksdl, 2017-11-04:
@tidyjiang8 牛！

blueliuyun, 2017-11-05:
@tidyjiang8 赞！

chengyi818, 2017-11-06:
@tidyjiang8 good job!

hanfengcan, 2017-11-18:
👍

closed 2017-12-19 by junjiec