alibaba/AliOS-Things #25: 关于BINS=app/framework/kernel，看不到具体的介绍啊

CrazySuxx, 2017-11-05:
sudo aos make helloworld@linuxhost是OK的
sudo aos make helloworld@linuxhost BINS=app, 这个命令有问题吗？还有BINS=framework和kernel分别要如何呢，这部分的文档没有呀

atmfjg, 2017-11-06:
欢迎使用多bin编译方式，现在开源版本支持kernel和app bin编译方式（两bin），后续版本会支持framework bin方式（三bin）。这里有介绍：
https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-BINS-Porting-Guide

tidyjiang8, 2017-11-06:
@atmfjg 
看文档，只介绍了一下如何编译多个 bin 文件，但是没说为啥需要编译多个 bin 文件/这样做的好处是啥

blueliuyun, 2017-11-06:
@tidyjiang8 多个 .bin 文件，实现了 kernel、APP 的组件化（组件化的方式弹性组合），从而比较方便调试&开发。eg. 某个 APP 应用代码的修改可以仅更新板子上的 App.bin，而板子上的内核 kernel.bin 则就可以不用一起更新； 同理，也适用于内核 bin 的更新操作。

blueliuyun, 2017-11-06:
```c
#ifdef MBEDTLS_IN_KERNEL
    PROVIDE(app_info_addr = 0x0070400);
    PROVIDE(kernel_download_addr = 0x0013200);
    PROVIDE(app_download_addr = 0x0077440);
    PROVIDE(ota_download_addr = 0x009c300);
#else
    PROVIDE(app_info_addr = 0x0067C00);
    PROVIDE(kernel_download_addr = 0x0013200);
    PROVIDE(app_download_addr = 0x006E3C0);
    PROVIDE(ota_download_addr = 0x00A5380);
#endif
```

@atmfjg  请问 多个 .bin 文件 和 单个 bin 方式下，**app_download_addr** 对应的加载地址为什么设计成不同的加载地址呢（*0x0077440*、*0x006E3C0*）？    我觉得在这里设计成同一个地址应该也可以吧。  向您请教这里设计成不同地址的考虑是什么呢？

atmfjg, 2017-11-06:
@tidyjiang8 @blueliuyun 面对物联网芯片资源有限的情况下，面对fota升级的硬性需求，除了查分算法等，我们是不是可以使用多bin方式来做这些，是不是可以节省相应的硬件flash资源。

atmfjg, 2017-11-06:
@blueliuyun 单bin不需要app_download_addr，这里有两份都是针对多bin的（理论上主要一份即可），一个针对mk3060的特殊需求。

blueliuyun, 2017-11-06:
@atmfjg 好的，谢谢回复。

closed 2017-12-26 by junjiec