alibaba/AliOS-Things #34: 编译之后display_map_summary失败

xzwang001, 2017-11-21:
按照例程编译helloworld，结果编译出现错误，二进制文件已经生成，且可以执行。但是编译还是存在错误，具体如下：
Build AOS Now
Traceback (most recent call last):
  File ".//build/scripts/map_parse_gcc.py", line 15, in <module>
    mem_config_text = re.findall('Memory Configuration\n\nName             Origin             Length             Attributes\n([\s\S]+)\nLinker script and memory map', s)[0]
IndexError: list index out of range
build/aos_target_build.mk:239: recipe for target 'display_map_summary' failed
make[1]: *** [display_map_summary] Error 1
build/Makefile:151: recipe for target 'main_app' failed

oujunli, 2017-11-24:
看一下out目录map文件内容，可能是语言等问题查找失败导致。可以提个PR @xzwang001

notagoodman, 2017-12-25:
哥们儿，你的问题解决了吗？我也遇到了同样的问题。

xzwang001, 2017-12-26:
@notagoodman，还没搞定呢，没有什么头绪

junjiec, 2017-12-26:
@xzwang001 @notagoodman 可以给一下完整的编译log吗，把debug打开，比如
aos make helloworld@linuxhost VERBOSE=1

xzwang001, 2017-12-26:
@junjiec ,以下信息是我这边编译时的log
`echo Build AOS Now
Build AOS Now
"/home/xzwang/Work/AliOS/AliOS-Things/build/cmd/linux64/mkdir" -p  out/helloworld@linuxhost/binary
"/home/xzwang/Work/AliOS/AliOS-Things/build/cmd/linux64/mkdir" -p  out/helloworld@linuxhost/modules
"/home/xzwang/Work/AliOS/AliOS-Things/build/cmd/linux64/mkdir" -p  out/helloworld@linuxhost/libraries
"/home/xzwang/Work/AliOS/AliOS-Things/build/cmd/linux64/mkdir" -p  out/helloworld@linuxhost/resources
"/home/xzwang/Work/AliOS/AliOS-Things/build/cmd/linux64/make" -r -j4  -f .//build/aos_target_build.mk helloworld@linuxhost  
make[1]: Entering directory '/home/xzwang/Work/AliOS/AliOS-Things'
"/usr/bin/python" .//build/scripts/map_parse_gcc.py out/helloworld@linuxhost/binary/helloworld@linuxhost.map     							
Traceback (most recent call last):
  File ".//build/scripts/map_parse_gcc.py", line 15, in <module>
    mem_config_text = re.findall('Memory Configuration\n\nName             Origin             Length             Attributes\n([\s\S]+)\nLinker script and memory map', s)[0]
IndexError: list index out of range
build/aos_target_build.mk:239: recipe for target 'display_map_summary' failed
make[1]: *** [display_map_summary] Error 1
make[1]: Leaving directory '/home/xzwang/Work/AliOS/AliOS-Things'
build/Makefile:151: recipe for target 'main_app' failed
make: *** [main_app] Error 2
[aos] ERROR: "/" returned error code 2.
[aos] ERROR: Command "/ h o m e / x z w a n g / W o r k / A l i O S / A l i O S - T h i n g s / b u i l d / c m d / l i n u x 6 4 / m a k e   H O S T _ O S = L i n u x 6 4   T O O L S _ R O O T = / h o m e / x z w a n g / W o r k / A l i O S / A l i O S - T h i n g s / b u i l d   - f   b u i l d / M a k e f i l e   h e l l o w o r l d @ l i n u x h o s t   V E R B O S E = 1" in "/home/xzwang/Work/AliOS/AliOS-Things"
---
`

xzwang001, 2017-12-26:
@junjiec ,以下信息是我这边打印的log

echo Build AOS Now
Build AOS Now
"/home/xzwang/Work/AliOS/AliOS-Things/build/cmd/linux64/mkdir" -p  out/helloworld@linuxhost/binary
"/home/xzwang/Work/AliOS/AliOS-Things/build/cmd/linux64/mkdir" -p  out/helloworld@linuxhost/modules
"/home/xzwang/Work/AliOS/AliOS-Things/build/cmd/linux64/mkdir" -p  out/helloworld@linuxhost/libraries
"/home/xzwang/Work/AliOS/AliOS-Things/build/cmd/linux64/mkdir" -p  out/helloworld@linuxhost/resources
"/home/xzwang/Work/AliOS/AliOS-Things/build/cmd/linux64/make" -r -j4  -f .//build/aos_target_build.mk helloworld@linuxhost  
make[1]: Entering directory '/home/xzwang/Work/AliOS/AliOS-Things'
"/usr/bin/python" .//build/scripts/map_parse_gcc.py out/helloworld@linuxhost/binary/helloworld@linuxhost.map     							
Traceback (most recent call last):
  File ".//build/scripts/map_parse_gcc.py", line 15, in <module>
    mem_config_text = re.findall('Memory Configuration\n\nName             Origin             Length             Attributes\n([\s\S]+)\nLinker script and memory map', s)[0]
IndexError: list index out of range
build/aos_target_build.mk:239: recipe for target 'display_map_summary' failed
make[1]: *** [display_map_summary] Error 1
make[1]: Leaving directory '/home/xzwang/Work/AliOS/AliOS-Things'
build/Makefile:151: recipe for target 'main_app' failed
make: *** [main_app] Error 2
[aos] ERROR: "/" returned error code 2.
[aos] ERROR: Command "/ h o m e / x z w a n g / W o r k / A l i O S / A l i O S - T h i n g s / b u i l d / c m d / l i n u x 6 4 / m a k e   H O S T _ O S = L i n u x 6 4   T O O L S _ R O O T = / h o m e / x z w a n g / W o r k / A l i O S / A l i O S - T h i n g s / b u i l d   - f   b u i l d / M a k e f i l e   h e l l o w o r l d @ l i n u x h o s t   V E R B O S E = 1" in "/home/xzwang/Work/AliOS/AliOS-Things"
---

junjiec, 2017-12-26:
@xzwang001 out/helloworld@linuxhost/binary/helloworld@linuxhost.elf 已经生成了？
out/helloworld@linuxhost/binary/helloworld@linuxhost.map 生成了吗？有的话发给我们看看

notagoodman, 2017-12-26:
@junjiec @xzwang001 
linuxhost.elf 没有生成。打印如下：
lxf@lxf-K43SA:/opt/AliOS-Things$ 
lxf@lxf-K43SA:/opt/AliOS-Things$ sudo aos make helloworld@linuxhost VERBOSE=1
[sudo] lxf 的密码： 
"/usr/bin/python" .//build/scripts/gen_test_collection.py out/helloworld@linuxhost/auto_component out/helloworld@linuxhost/auto_component/test_collection.default
"/usr/bin/python" .//build/scripts/auto_component.py out/helloworld@linuxhost/auto_component
echo Making config file for first time
Making config file for first time
"/opt/AliOS-Things/build/cmd/linux64/make" -r  -f .//build/aos_target_config.mk helloworld@linuxhost
make[1]: Entering directory '/opt/AliOS-Things'
processing components: helloworld linuxhost platform/mcu/linux vcall init auto_component
kernel_version:AOS-R-1.1.2
echo 'AOS_SDK_MAKEFILES           		+= ././kernel/vfs/device/device.mk ./security/alicrypto/alicrypto.mk ./kernel/hal/hal.mk ./kernel/vfs/vfs.mk ./kernel/rhino/rhino.mk ./security/mbedtls/mbedtls.mk ./utility/log/log.mk ./security/tfs/tfs.mk ./tools/cli/cli.mk ./kernel/yloop/yloop.mk ./out/helloworld@linuxhost/auto_component/auto_component.mk ./kernel/init/init.mk ./kernel/vcall/vcall.mk ././platform/mcu/linux/linux.mk ./board/linuxhost/linuxhost.mk ./example/helloworld/helloworld.mk' >  out/helloworld@linuxhost/config.mk ;
echo 'TOOLCHAIN_NAME            		:= GCC' >>  out/helloworld@linuxhost/config.mk ;
echo 'AOS_SDK_LDFLAGS             		+= -Wl,--gc-sections -Wl,--cref -lpthread -lm -lrt -m32' >>  out/helloworld@linuxhost/config.mk ;
echo 'AOS_SDK_LDS_FILES                     += ' >>  out/helloworld@linuxhost/config.mk ;
echo 'AOS_SDK_LDS_INCLUDES                  += ' >>  out/helloworld@linuxhost/config.mk ;
echo 'RESOURCE_CFLAGS					+= -DSYSINFO_KERNEL_VERSION=\\"AOS-R-1.1.2\\" -Wall -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-address -Wno-unused-result -DSYSINFO_PRODUCT_MODEL=\\"ALI_AOS_LINUXHOST\\" -DSYSINFO_DEVICE_NAME=\\"LINUXHOST\\" -m32 -std=gnu99' >>  out/helloworld@linuxhost/config.mk ;
echo 'AOS_SDK_LINK_SCRIPT         		+= ' >>  out/helloworld@linuxhost/config.mk ;
echo 'AOS_SDK_LINK_SCRIPT_CMD    	 	+= ' >>  out/helloworld@linuxhost/config.mk ;
echo 'AOS_SDK_PREBUILT_LIBRARIES 	 	+= ./security/tfs/lib/linux/libtfs.a ./security/tfs/lib/linux/libtfspal.a ./security/mbedtls/lib/linux/libmbedtls.a ./security/alicrypto/lib/linux/libmbedcrypto.a ./security/alicrypto/lib/linux/libalicrypto.a' >>  out/helloworld@linuxhost/config.mk ;
echo 'AOS_SDK_CERTIFICATES       	 	+= ' >>  out/helloworld@linuxhost/config.mk ;
echo 'AOS_SDK_PRE_APP_BUILDS      		+= ' >>  out/helloworld@linuxhost/config.mk ;
echo 'AOS_SDK_LINK_FILES          		+=                                ' >>  out/helloworld@linuxhost/config.mk ;
echo 'AOS_SDK_INCLUDES           	 	+=                 -I./security/alicrypto/inc -I./security/alicrypto/test -I./kernel/vfs/include -I./kernel/rhino/core/include -I./security/mbedtls/include -I./security/tfs/include -I./tools/cli/include -I./kernel/vcall/./mico/include -I././platform/mcu/linux/. -I././platform/mcu/linux/../../arch/linux/ -I././platform/mcu/linux/include -I././platform/mcu/linux/include/aos -I././platform/mcu/linux/csp/lwip/include -I./board/linuxhost/. -I./include -I./example/helloworld' >>  out/helloworld@linuxhost/config.mk ;
echo 'AOS_SDK_DEFINES             		+=                 -DBUILD_BIN -DCONFIG_ALICRYPTO -DAOS_HAL -DAOS_VFS -DHAVE_NOT_ADVANCED_FORMATE -DCONFIG_AOS_CLI -DAOS_LOOP -DVCALL_RHINO -DCONFIG_AOS_RHINO_MMREGION -DCONFIG_YSH_CMD_DUMPSYS -DCSP_LINUXHOST -DCONFIG_LOGMACRO_DETAILS -DCONFIG_AOS_FATFS_SUPPORT -DCONFIG_AOS_FATFS_SUPPORT_MMC -DCONFIG_AOS_FOTA_BREAKPOINT -DAOS_NO_WIFI' >>  out/helloworld@linuxhost/config.mk ;
echo 'COMPONENTS                		:= helloworld board_linuxhost linuximpl vcall kernel_init auto_component yloop cli libtfs.a log mbedtls rhino vfs hal alicrypto vfs_device' >>  out/helloworld@linuxhost/config.mk ;
echo 'PLATFORM_DIRECTORY        		:= linuxhost' >>  out/helloworld@linuxhost/config.mk ;
echo 'APP_FULL                  		:= helloworld' >>  out/helloworld@linuxhost/config.mk ;
echo 'PLATFORM                  		:= linuxhost' >>  out/helloworld@linuxhost/config.mk ;
echo 'HOST_MCU_FAMILY                  	:= linux' >>  out/helloworld@linuxhost/config.mk ;
echo 'SUPPORT_BINS                          := no' >>  out/helloworld@linuxhost/config.mk ;
echo 'APP                       		:= helloworld' >>  out/helloworld@linuxhost/config.mk ;
echo 'HOST_OPENOCD              		:= linux' >>  out/helloworld@linuxhost/config.mk ;
echo 'JTAG              		        := jlink_swd' >>  out/helloworld@linuxhost/config.mk ;
echo 'HOST_ARCH                 		:= linux' >>  out/helloworld@linuxhost/config.mk ;
echo 'NO_BUILD_BOOTLOADER           	:= ' >>  out/helloworld@linuxhost/config.mk ;
echo 'NO_BOOTLOADER_REQUIRED         	:= ' >>  out/helloworld@linuxhost/config.mk ;
echo 'helloworld_LOCATION         := ./example/helloworld/' >>  out/helloworld@linuxhost/config.mk ;  echo 'board_linuxhost_LOCATION         := ./board/linuxhost/' >>  out/helloworld@linuxhost/config.mk ;  echo 'linuximpl_LOCATION         := ././platform/mcu/linux/' >>  out/helloworld@linuxhost/config.mk ;  echo 'vcall_LOCATION         := ./kernel/vcall/' >>  out/helloworld@linuxhost/config.mk ;  echo 'kernel_init_LOCATION         := ./kernel/init/' >>  out/helloworld@linuxhost/config.mk ;  echo 'auto_component_LOCATION         := ./out/helloworld@linuxhost/auto_component/' >>  out/helloworld@linuxhost/config.mk ;  [... cut ...]
make[1]: Leaving directory '/opt/AliOS-Things'
build/Makefile:197: recipe for target 'main_app' failed
make: *** [main_app] Error 2
[aos] ERROR: "/" returned error code 2.
[aos] ERROR: Command "/ o p t / A l i O S - T h i n g s / b u i l d / c m d / l i n u x 6 4 / m a k e   H O S T _ O S = L i n u x 6 4   T O O L S _ R O O T = / o p t / A l i O S - T h i n g s / b u i l d   - f   b u i l d / M a k e f i l e   h e l l o w o r l d @ l i n u x h o s t   V E R B O S E = 1" in "/opt/AliOS-Things"
---

notagoodman, 2017-12-26:
@junjiec 纠正下，out/helloworld@linuxhost/binary/helloworld@linuxhost.elf已经生成了。map_parse_gcc.py貌似有问题，我没有改过，按照编译的提示，其中第15行如下：
 mem_config_text = re.findall('Memory Configuration\n\nName             Origin             Length             Attributes\n([\s\S]+)\nLinker script and memory map', s)[0]

[helloworld@linuxhost.map.zip](https://github.com/alibaba/AliOS-Things/files/1587195/helloworld.linuxhost.map.zip)

junjiec, 2017-12-26:
@notagoodman 最好不要用sudo，把out目录删除，不用sudo编译看看

notagoodman, 2017-12-26:
@junjiec 试过，不行。每次编译之前都会手动删除out目录。
我建立的目录如果不用sudo就没有权限，所以加了sudo。
之前在个人目录下，没有使用sudo可以编译，但也是一样的结果。

junjiec, 2017-12-26:
@notagoodman 是语言设置的问题，设置为中文会有问题，编译前
export LANG=en_US
一下试试

notagoodman, 2017-12-26:
@junjiec 厉害，就是语言设置问题。重新设置了语言后就通过了。
请教下，你从哪里看出来是语言问题的？

junjiec, 2017-12-26:
@oujunli 同学发现的,map文件里面有中文字符

closed 2017-12-26 by junjiec