alibaba/AliOS-Things #37: AliOS Things 支持C++开发吗

KaiboFu, 2017-11-24:


junjiec, 2017-11-24:
目前不支持

KaiboFu, 2017-11-26:
感谢 希望尽快支持

librae8226, 2017-11-28:
@KaiboFu 想了解下你是希望用C++来做哪方面的东西呢？是想要运行一些类似Wiring的代码吗？

KaiboFu, 2017-11-28:
@librae8226 是的，感觉Arduino式的编程风格简单且易理解。（我刚接触到一些嵌入书开发的东西，问的问题如果有些low，还望谅解）
其实主要是C++语言本身的一些特性，主要是封装，继承两点。下午试了一下MiCO，发现其本身是支持C++开发的（MiCO自身提供的示例：https://code.aliyun.com/mico/drv_DHT11.git 对dht11进行一些封装），所以AliOS-Things理论上支持C++开发是不是只需要变换一下交叉编译链就可以了呢？
（附加一个问题： 如果想用AliOS-Things对MK3060上的一些硬件接口及设备进行开发，例如GPIO、ADC、I2C和其扩展板上自带的传感器等，是不是直接运用MiCO提供的一些API就可以了呢？AliOS有没有相关一些示例代码和相关文档呢？）

junjiec, 2017-11-29:
@KaiboFu 如果不用到c++的一些动态高级特性RTTI，exception等，那支持起来应该比较快(主要是编译支持及系统初始化时调用c++的constructor)，欢迎提交补丁:)。

librae8226, 2017-12-15:
我简单试了下，如 @junjiec 提到，如果不需要c++高级特性，默认应该可以编译出来运行的
需要的话可以参考一下，目前可以支持 esp32 和 mk3060
https://github.com/librae8226/AliOS-Things/commit/a2cff3b2bf090f8de1a216550358e31822e5218a

KaiboFu, 2017-12-17:
@librae8226 感谢 确实和猜想的一样

closed 2017-12-18 by junjiec