alibaba/AliOS-Things #40: 希望弄重写下部分说明

aiwonanjing, 2017-11-30:
第一次看到这个项目主页时一脸懵逼。原因很简单，主页上面没有开发适用平台（核心架构）和系统基本使用方法。希望在这个问题上参考下[https://github.com/RT-Thread/rt-thread](url)，下载链接并不是系统镜像而是一大堆其他软件的下载链接，而这些链接放在系统开发的方法中比较好吧。
文档排版也相当混乱，一会儿是中文，一会儿是英文，如果考虑国际化推广，务必考虑双语种。

junjiec, 2017-11-30:
多谢建议

aiwonanjing, 2018-02-08:
个人对RTOS的运行模式比较好奇，却发现wiki中好像没有类似于https://www.rt-thread.org/document/site/figures/System_Arch.png这样的系统结构，是否可以在wiki中补充一下？既然开源了，也可以考虑让各位大神了解系统的运作模式，这样也便于系统后期的优化发展。
我只是个爱好者，叙述可能不清楚。

junjiec, 2018-02-08:
@aiwonanjing 类似于这个？
https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-Technical-Overview#block-diagram

aiwonanjing, 2018-02-08:
抱歉，没注意到，可能是位置不够显眼造成不好找的。
可以在项目readme上把wiki直接链接上去。

aiwonanjing, 2018-02-10:
Technical Overview被扔到了Links里面，太不显眼了，wiki的首页还可以细化修改下，而且wiki在readme的位置应该明显点。

librae8226, 2018-03-02:
github wiki 上的文档脉络已经过一轮整理
https://github.com/alibaba/AliOS-Things/wiki
感谢提出宝贵意见

closed 2022-03-01 by skylarCai