alibaba/AliOS-Things #53: AliOS-Things 在MAC上的配置

dupig, 2017-12-21:
作为一个搞交互设计的小白，对AliOS-Things的应用前景感到很兴奋，很想利用它做一些场景性的探索。于是花了一点时间和怡然讨论整理出了AliOS-Things在MAC上的配置，供大家参考。MAC跟Windows的步骤其实差不多，主要是一些安装包的配置上会有些区别。
1. 首先下载visual studio code，然后安装
[image]
2. 安装之后打开，点击红圈进入安装扩展页面
[image]
3. 在应用搜索中输入alios，唯一一个的扩展就是它
[image]
4. 然后下载安装alios-studio，需要重启一次VS Code。重启后扩展生效，回到欢迎页面点击右下角create project，或者用键盘command+P打开命令窗口输入“>alios-studio：Create project”
[image]
[image]
5. 第一次创立项目的时候需要配置SDK，注意这时对比windows的界面是多了一个tool-chain path的
[image]
6. SDK path需要从GitHub上下载alios-things的[开发包](https://github.com/alibaba/AliOS-Things)
[image]
7. 我个人把它放到VS Code的extension目录下然后解压，注意这个.vscode是隐藏的，需要按command+shift+. 显示
[image]
8. 除此之外还需要下载[交叉工具链](https://launchpad.net/gcc-arm-embedded/+download)
[image]
9. 下载解压后复制路径到tool-chain path里面
[image]
然后就可以create project了，之后流程跟windows一样了。

librae8226, 2017-12-21:
🍻 Thanks! @dupig

junjiec, 2017-12-21:
@dupig 感谢，已放到wiki
https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-MAC-Environment-Setup

closed 2017-12-21 by junjiec