alibaba/AliOS-Things #55: [AliOS-studio][ESP32] tool chain path.

jackon, 2017-12-22:
设了 esp32 的 tool chain path，一直不会生效，无休止的跳出 “The Path is required!”，必须要设成 arm 的。难道内部只校验了 arm tool chain 的 prefix？

librae8226, 2017-12-27:
抱歉带来的不便
我也有遇到，你可以临时这样 hack 一下 😅
因为 AliOS-Studio 在快速迭代中，这个问题关系到一些设计上的完善，等正式 fix 以后我们再来 close 该 issue
```
diff --git a/src/yang-sdk/main.js b/src/yang-sdk/main.js
index a41520e..42c1582 100644
--- a/src/yang-sdk/main.js
+++ b/src/yang-sdk/main.js
@@ -24,9 +24,9 @@ module.exports = {
         if (!fsPlus.isDirectorySync(dir)) {
             return {error: 1, msg: 'path is not a directory!'};
         }
-        if (options.toolChain && !fsPlus.existsSync(path.join(dir, `arm-none-eabi-gcc${process.platform === 'win32' ? '.exe' : ''}`))) {
-            return {error: 1, msg: 'path is not a tool chain directory!'};
-        }
+//      if (options.toolChain && !fsPlus.existsSync(path.join(dir, `arm-none-eabi-gcc${process.platform === 'win32' ? '.exe' : ''}`))) {
+//          return {error: 1, msg: 'path is not a tool chain directory!'};
+//      }
         if (options.version) {
             let filePath = path.join(dir, 'kernel', 'rhino', 'rhino.mk'),
                 pkg = fsPlus.existsSync(filePath) && fs.readFileSync(filePath, {encoding: 'utf-8'}),
```

junjiec, 2018-01-12:
最新版本IDE已经修复

closed 2018-01-12 by junjiec