alibaba/AliOS-Things #74: ESP32 libdriver.a 库文件 FreeRTOS 接口缺失

shaoguoji, 2018-01-13:
在封装乐鑫提供的 GPIO 接口 .\platform\mcu\esp32\bsp\include\driver\include\driver\gpio.h 中断相关的函数时（gpio_isr_handler_add、gpio_isr_handler_remove、gpio_install_isr_service、gpio_uninstall_isr_service），编译报错找不到 vTaskEnterCritical 和 vTaskExitCritical 两个与临界区相关的操作，进一步得知两者来源于 esp-idf 中 FreeRTOS 组件中， aos 似乎没有将两者编译到目标 \platform\mcu\esp32\lib\libdriver.a 中，也没有提供 FreeRTOS 组件。

`[ERROR-Build] ././platform/mcu
[ERROR-Build] /esp32/lib/libdriver.a(gpio.o):(.literal.gpio_isr_handler_add+0x14): undefined reference to `vTaskEnterCritical'
[ERROR-Build] ././platform/mcu/esp32/lib/libdriver.a(gpio.o):(.literal.gpio_isr_handler_add+0x18): undefined reference to `vTaskExitCritical'
[ERROR-Build] ././platform/mcu/esp32/lib/libdriver.a(gpio.o): In function `gpio_isr_handler_add':
[ERROR-Build] /home/junjie-cai/github/esp-idf-alios/components/driver/./gpio.c:428: undefined reference to `vTaskEnterCritical'
[ERROR-Build] /home/junjie-cai/github/esp-idf-alios/components/driver/./gpio.c:428: undefined reference to `vTaskExitCritical'
[ERROR-Build] ././platform/mcu/esp32/lib/libdriver.a(gpio.o): In function `gpio_isr_handler_remove':
[ERROR-Build] /home/junjie-cai/github/esp-idf-alios/components/driver/./gpio.c:428: undefined reference to `vTaskEnterCritical'
[ERROR-Build] /home/junjie-cai/github/esp-idf-alios/components/driver/./gpio.c:428: undefined reference to `vTaskExitCritical'
[ERROR-Build] ././platform/mcu/esp32/lib/libdriver.a(gpio.o): In function `gpio_install_isr_service':
[ERROR-Build] /home/junjie-cai/github/esp
[ERROR-Build] -idf-alios/components/driver/./gpio.c:406: undefined reference to `vTaskEnterCritical'
[ERROR-Build] /home/junjie-cai/github/esp-idf-alios/components/driver/./gpio.c:413: undefined reference to `vTaskExitCritical
[ERROR-Build] '
[ERROR-Build] collect2.exe: error: ld returned 1 exit status`

尝试通过更底层操作重写中断相关函数，但程序无法正常运行（无效果、跑飞重启），请问该如何解决此问题从而调用乐鑫 gpio 接口中断相关函数，能力有限，感谢解答……

junjiec, 2018-01-15:
收到，我们评估一下怎么提供相关资源

junjiec, 2018-01-16:
@shaoguoji libdriver.a已更新

closed 2018-01-16 by junjiec