alibaba/AliOS-Things #103: 编译问题

LShaoPeng, 2018-02-03:
编译过程中出现下面的log，然后就没法继续编译了，重新启动进行编译会出现gcc相关的找不到文件等错误。
[INFO-Build] -------------OUTPUT START-------------
[INFO-Build] Making config file for first time
[INFO-Build] 
[INFO-Build] processing components: jingshuiqi mk3060 platform/mcu/moc108 vcall init auto_component
[INFO-Build]

junjiec, 2018-02-03:
给一下更具体的命令,把VERBOSE打开，比如
aos make helloworld@linuxhost VERBOSE=1

LShaoPeng, 2018-02-03:
这个问题我刚刚尝试解决了一下，我将.mk文件替换成最新的之后，编译就没有问题了，编译命令：aos make alinkapp@mk3060 sds=1

zhuoshuguo, 2018-02-08:
问一个小白问题，按照教程（包括视频教程），Windows平台，下完VS code后，创建helloworld工程后，点击build，为啥老是（找不到路径？）：
```
[INFO-Build] Building...
[INFO-Build] -------------COMPILE OUTPUT START-------------
[INFO-Build] Making config file for first time
[INFO-Build] 
[INFO-Build] processing components: helloworld mk3060 platform/mcu/moc108 vcall init auto_component
[INFO-Build] 
[INFO-Build] Unknown component: helloworld - directory or makefile for component not found. Ensure the helloworld directory contains helloworld.mk
[INFO-Build] Below is a list of valid local components (Some are internal):
[INFO-Build] 
[INFO-Build] mcu.mkl27z644
[INFO-Build] mcu.csky
....
```

我错过了什么么？

zhuoshuguo, 2018-02-08:
但是在Alios-things的源码目录下运行：
```
$ aos make helloworld@mk3060
```
貌似是可以编译的。编译成功。如何在VS code 中编译成功？

librae8226, 2018-03-02:
感谢反馈，该问题已于近期修复，参考:
fb98c0a
> fix bug when set APPDIR component

zhuoshuguo, 2018-03-02:
非常感谢！~这回可以编译通过~

librae8226, 2018-03-02:
👍cool

closed 2018-03-02 by librae8226