alibaba/AliOS-Things #115: cotexm3支持吗？看前面的问题知道arch和m4公用就行了但MCU那部分怎么解决？要支持m3怎么改？

supeng6751, 2018-02-08:


chenlingjun888, 2018-02-08:
可以参照现有的cortex-m4的那块，加入 stm32的固件库，包含启动代码，就可以了。

supeng6751, 2018-02-09:
感谢回答，昨天将st的hal库以及启动代码移植了我用的是stm32l1系列mcu，想使用的是lorawan的那个工程，但仍然报了大量错误，我仔细看了一下，是因为目前使用的mcu使用的hal库是老版本的库而stm32l1使用的 是cube1的hal库，hal库不一致导致的，两个库不兼容导致board里会报出大量错误，如果这样的话我目前有两种思路1、在我们原来的工程基础上移植alios。2、改写不兼容的函数。目前使用的是方法2，这两个思路有没有问题？如果思路没问题的话那个方法更优？如果都有问题有没有移植的建议？

supeng6751, 2018-02-09:
@chenlingjun888  感谢回答，昨天将st的hal库以及启动代码移植了我用的是stm32l1系列mcu，想使用的是lorawan的那个工程，但仍然报了大量错误，我仔细看了一下，是因为目前使用的mcu使用的hal库是老版本的库而stm32l1使用的 是cube1的hal库，hal库不一致导致的，两个库不兼容导致board里会报出大量错误，如果这样的话我目前有两种思路1、在我们原来的工程基础上移植alios。2、改写不兼容的函数。目前使用的是方法2，这两个思路有没有问题？如果思路没问题的话那个方法更优？如果都有问题有没有移植的建议？

shengpingdai, 2018-02-24:
两种思路都可以，不兼容的函数多吗？如果不多你可以继续使用方法2尽快实现，如果多的话还是使用思路1比较好。个人觉得不改底层的库比较好，因为你别的工程也可能会用到这些库，改动之后可能会造成别的工程出问题。

junjiec, 2018-04-27:
最新版本已经支持:platform/arch/arm/armv7m/gcc/m3/

closed 2018-04-27 by junjiec