alibaba/AliOS-Things #124: FreeRTOS相关文件缺失

wanglingsong, 2018-02-10:
我正在尝试在aos项目里使用esp32的rmt模组，但编译的时候出现如下错误：
```
././platform/mcu/esp32/bsp/include/driver/include/driver/rmt.h:21:31: fatal error: freertos/FreeRTOS.h: No such file or directory
```
请问应该如何恰当的引入这项freertos相关文件？

junjiec, 2018-02-11:
先把头文件里面freertos相关的头文件去掉，往下走，驱动也需要一些相应的修改，这个我们后面放上去

Ever-Never, 2018-02-11:
就是这个错误  我已经自己使用自己工程的FreeRTOS的头文件已经跳过这步骤了

wanglingsong, 2018-02-11:
我注释掉了相关的头文件也修改了驱动，编译后出现如下错误：
```
Making iotlink@esp32devkitc.elf
././platform/mcu/esp32/lib/libdriver.a(rmt.o):(.iram1.literal+0xc): undefined reference to `vTaskEnterCritical'
././platform/mcu/esp32/lib/libdriver.a(rmt.o):(.iram1.literal+0x10): undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(rmt.o):(.iram1.literal+0x30): undefined reference to `xQueueGiveFromISR'
././platform/mcu/esp32/lib/libdriver.a(rmt.o):(.literal.rmt_set_tx_wrap_en+0x8): undefined reference to `vTaskEnterCritical'
././platform/mcu/esp32/lib/libdriver.a(rmt.o):(.literal.rmt_set_tx_wrap_en+0xc): undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(rmt.o):(.literal.rmt_driver_install+0x1c): undefined reference to `xQueueGenericCreate'
././platform/mcu/esp32/lib/libdriver.a(rmt.o):(.literal.rmt_driver_install+0x20): undefined reference to `xQueueGenericSend'
././platform/mcu/esp32/lib/libdriver.a(rmt.o):(.literal.rmt_write_items+0x10): undefined reference to `xQueueGenericReceive'
././platform/mcu/esp32/lib/libdriver.a(rmt.o): In function `rmt_fill_memory':
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(rmt.o): In function `rmt_driver_isr_default':
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `xQueueGiveFromISR'
././platform/mcu/esp32/lib/libdriver.a(rmt.o): In function `rmt_set_tx_wrap_en':
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(rmt.o): In function `rmt_set_data_mode':
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(rmt.o): In function `rmt_tx_start':
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(rmt.o): In function `rmt_set_intr_enable_mask':
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(rmt.o): In function `rmt_clr_intr_enable_mask':
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(rmt.o): In function `rmt_config':
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskExitCritical'
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskExitCritical'
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./rmt.c:661: undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(rmt.o): In function `rmt_driver_install':
/home/cai/esp-idf-alios/components/driver/./rmt.c:690: undefined reference to `xQueueGenericCreate'
/home/cai/esp-idf-alios/components/driver/./rmt.c:691: undefined reference to `xQueueGenericSend'
././platform/mcu/esp32/lib/libdriver.a(rmt.o): In function `rmt_write_items':
/home/cai/esp-idf-alios/components/driver/./rmt.c:725: undefined reference to `xQueueGenericReceive'
/home/cai/esp-idf-alios/components/driver/./rmt.c:745: undefined reference to `xQueueGenericReceive'
/home/cai/esp-idf-alios/components/driver/./rmt.c:746: undefined reference to `xQueueGenericSend'
collect2.exe: error: ld returned 1 exit status
make.exe[1]: *** [build/aos_target_build.mk:286: out/iotlink@esp32devkitc/binary/iotlink@esp32devkitc.elf] Error 1
make.exe: *** [build/Makefile:219: main_app] Error 2
[AliOS-Things] ERROR: "D" returned error code 2.
```

junjiec, 2018-02-12:
尝试一下esp32_driver这个分支，rmt／ledc都修了一下

wanglingsong, 2018-02-12:
rmt已经能编译通过了。感谢！

closed 2018-02-12 by wanglingsong