alibaba/AliOS-Things #125: LD error

Ever-Never, 2018-02-11:
/home/fine/aos/build/compiler/arm-none-eabi-5_4-2016q3-20160926/Linux64/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: error: ././platform/mcu/mt2523/project/mt2523_hdk/apps/iot_sdk_demo/lib/iot.a(deflate.o) uses VFP register arguments, out/helloworld@mt2523/binary/helloworld@mt2523.elf does not
/home/fine/aos/build/compiler/arm-none-eabi-5_4-2016q3-20160926/Linux64/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file ././platform/mcu/mt2523/project/mt2523_hdk/apps/iot_sdk_demo/lib/iot.a(deflate.o)
/home/fine/aos/build/compiler/arm-none-eabi-5_4-2016q3-20160926/Linux64/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: error: ././platform/mcu/mt2523/project/mt2523_hdk/apps/iot_sdk_demo/lib/iot.a(infback.o) uses VFP register arguments, out/helloworld@mt2523/binary/helloworld@mt2523.elf does not
/home/fine/aos/build/compiler/arm-none-eabi-5_4-2016q3-20160926/Linux64/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file ././platform/mcu/mt2523/project/mt2523_hdk/apps/iot_sdk_demo/lib/iot.a(infback.o)
/home/fine/aos/build/compiler/arm-none-eabi-5_4-2016q3-20160926/Linux64/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: error: ././platform/mcu/mt2523/project/mt2523_hdk/apps/iot_sdk_demo/lib/iot.a(inflate.o) uses VFP register arguments, out/helloworld@mt2523/binary/helloworld@mt2523.elf does not
/home/fine/aos/build/compiler/arm-none-eabi-5_4-2016q3-20160926/Linux64/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file ././platform/mcu/mt2523/project/mt2523_hdk/apps/iot_sdk_demo/lib/iot.a(inflate.o)
/home/fine/aos/build/compiler/arm-none-eabi-5_4-2016q3-20160926/Linux64/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: error: ././platform/mcu/mt2523/project/mt2523_hdk/apps/iot_sdk_demo/lib/iot.a(inftrees.o) uses VFP register arguments, out/helloworld@mt2523/binary/helloworld@mt2523.elf does not

Ever-Never, 2018-02-11:
配置自己交叉编译器 ?   $(TOOLCHAIN_PATH)$(TOOLCHAIN_PREFIX)

librae8226, 2018-05-23:
硬浮点的编译选项 `-mfloat-abi` 需要统一，如果把不同编译选项编出来的目标链接到一起，就会报这个问题

closed 2018-05-23 by librae8226