alibaba/AliOS-Things #126: ESP32  接口缺失

wanglingsong, 2018-02-11:
我正在尝试用esp32的ledc组件实现esp32的pwm hal接口。编译是出现如下错误。
与这个issue情况类似：
https://github.com/alibaba/AliOS-Things/issues/74
```
Making iotlink@esp32devkitc.elf
././platform/mcu/esp32/lib/libdriver.a(ledc.o):(.iram1.literal+0x18): undefined reference to `vTaskEnterCritical'
././platform/mcu/esp32/lib/libdriver.a(ledc.o):(.iram1.literal+0x1c): undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(ledc.o):(.iram1.literal+0x28): undefined reference to `xQueueGiveFromISR'
././platform/mcu/esp32/lib/libdriver.a(ledc.o):(.literal.ledc_enable_intr_type+0x1c): undefined reference to `vTaskEnter
Critical'
././platform/mcu/esp32/lib/libdriver.a(ledc.o):(.literal.ledc_enable_intr_type+0x20): undefined reference to `vTaskExitC
ritical'
././platform/mcu/esp32/lib/libdriver.a(ledc.o): In function `ledc_duty_config':
/home/cai/esp-idf-alios/components/driver/./ledc.c:334: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./ledc.c:334: undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(ledc.o): In function `ledc_fade_isr':
/home/cai/esp-idf-alios/components/driver/./ledc.c:334: undefined reference to `xQueueGiveFromISR'
././platform/mcu/esp32/lib/libdriver.a(ledc.o): In function `ledc_enable_intr_type':
/home/cai/esp-idf-alios/components/driver/./ledc.c:334: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./ledc.c:334: undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(ledc.o): In function `ledc_timer_set':
/home/cai/esp-idf-alios/components/driver/./ledc.c:334: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./ledc.c:334: undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(ledc.o): In function `ledc_bind_channel_timer':
/home/cai/esp-idf-alios/components/driver/./ledc.c:334: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./ledc.c:334: undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(ledc.o): In function `ledc_timer_rst':
/home/cai/esp-idf-alios/components/driver/./ledc.c:334: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./ledc.c:334: undefined reference to `vTaskExitCritical'
././platform/mcu/esp32/lib/libdriver.a(ledc.o): In function `ledc_update_duty':
/home/cai/esp-idf-alios/components/driver/./ledc.c:334: undefined reference to `vTaskEnterCritical'
/home/cai/esp-idf-alios/components/driver/./ledc.c:334: undefined reference to `vTaskExitCritical'
collect2.exe: error: ld returned 1 exit status
make.exe[1]: *** [build/aos_target_build.mk:286: out/iotlink@esp32devkitc/binary/iotlink@esp32devkitc.elf] Error 1
make.exe: *** [build/Makefile:219: main_app] Error 2
[AliOS-Things] ERROR: "D" returned error code 2.
```

Ever-Never, 2018-02-11:
free rtos api 换成alios api 就能编译过了

wanglingsong, 2018-02-11:
@Ever-Never 你好请问具体怎么换？

wanglingsong, 2018-02-12:
#124

junjiec, 2018-02-12:
已修改 @wanglingsong

wanglingsong, 2018-02-12:
编译成功

closed 2018-02-12 by wanglingsong