alibaba/AliOS-Things #139: 在vscode中编译hellowrold@esp32devkitc失败

flyingcys, 2018-02-24:
希望给予一些解决问题的帮助！谢谢
错误提示如下：
[INFO-Build] Building...
[INFO-Build] -------------COMPILE OUTPUT START-------------
[INFO-Build] Making config file for first time
[INFO-Build] 
[INFO-Build] processing components: helloworld esp32devkitc platform/mcu/esp32 vcall init auto_component
[INFO-Build] 
[INFO-Build] Unknown component: helloworld - directory or makefile for component not found. Ensure the helloworld directory contains helloworld.mk
[INFO-Build] Below is a list of valid local components (Some are internal):
[INFO-Build] 
[INFO-Build] mcu.mkl27z644
[INFO-Build] mcu.rtl8710bn
[INFO-Build] mcu.rtl8710bn.sdk
[INFO-Build] mcu.rtl8710bn.peripherals
[INFO-Build] mcu.csky
[INFO-Build] mcu.csky.hal_init
[INFO-Build] mcu.mx1101
[INFO-Build] mcu.mx1101.sdk
[INFO-Build] mcu.esp32
[INFO-Build] 
[INFO-Build] mcu.lpc54102
[INFO-Build] mcu.esp8266
[INFO-Build] mcu.linux
[INFO-Build] mcu.stm32l475
[INFO-Build] mcu.stm32f4xx
[INFO-Build] mcu.stm32f4xx.spi_flash
[INFO-Build] mcu.stm32f4xx.peripherals
[INFO-Build] mcu.stm32f4xx.peripherals.Libraries
[INFO-Build] mcu.stm32l0xx.stm32l071kb
[INFO-Build] mcu.moc108
[INFO-Build] mcu.moc108.aos.app_runtime
[INFO-Build] mcu.moc108.aos.framework_runtime
[INFO-Build] mcu.xr871
[INFO-Build] mcu.xr871.lib.libamr
[INFO-Build] mcu.xr871.lib.libmp3
[INFO-Build] mcu.xr871.lib.libamren
[INFO-Build] mcu.xr871.src.net.wlan
[INFO-Build] mcu.xr871.src.net.lwip
[INFO-Build] mcu.xr871.src.net.udhcp
[INFO-Build] mcu.xr871.src.efpg
[INFO-Build] mcu.xr871.src.driver.chip
[INFO-Build] mcu.xr871.src.image
[INFO-Build] mcu.xr871.src.ota
[INFO-Build] mcu.xr871.src.sys
[INFO-Build] mcu.xr871.src.console
[INFO-Build] mcu.xr871.src.audio.audio_manager
[INFO-Build] mcu.xr871.src.audio.audio_pcm
[INFO-Build] mcu.xr871.src.pm
[INFO-Build] mcu.xr871.src.cedarx
[INFO-Build] mcu.xr871.src.xz
[INFO-Build] mcu.xr871.project
[INFO-Build] mcu.xr871.aos
[INFO-Build] arch.csky.cskyv2-l
[INFO-Build] arch.xtensa.lx6
[INFO-Build] arch.xtensa.lx106
[INFO-Build] arch.linux
[INFO-Build] arch.arm.armv7m
[INFO-Build] arch.arm.armv6m
[INFO-Build] arch.arm.armv5
[INFO-Build] init
[INFO-Build] modules.fs.kv
[INFO-Build] modules.fs.spiffs
[INFO-Build] modules.fs.uffs
[INFO-Build] modules.fs.fatfs
[INFO-Build] vfs
[INFO-Build] vfs.device
[INFO-Build] hal
[INFO-Build] yloop
[INFO-Build] vcall
[INFO-Build] vcall.espos
[INFO-Build] ksyscall
[INFO-Build] rhino
[INFO-Build] protocols.net
[INFO-Build] protocols.mesh
[INFO-Build] protocols.bluetooth
[INFO-Build] protocols.bluetooth.host.bt_mesh
[INFO-Build] protocols.lorawan
[INFO-Build] opus
[INFO-Build] opus.celt
[INFO-Build] opus.silk
[INFO-Build] cloud
[INFO-Build] protocol.alink
[INFO-Build] protocol.alink.accs
[INFO-Build] protocol.alink.cota
[INFO-Build] protocol.alink.json
[INFO-Build] protocol.alink.os
[INFO-Build] protocol.alink.system
[INFO-Build] netmgr
[INFO-Build] fota
[INFO-Build] fota.platform
[INFO-Build] fota.platform.mqtt
[INFO-Build] fota.platform.coap
[INFO-Build] fota.platform.alink
[INFO-Build] fota.platform.common
[INFO-Build] fota.download
[INFO-Build] fota.download.http
[INFO-Build] fota.download.http.socket
[INFO-Build] 
[ERROR-Build] build/aos_target_config.mk:311: *** Unknown component: helloworld - directory or makefile for component not found. Ensure the helloworld directory contains helloworld.mk.  Stop.
[ERROR-Build] 
[INFO-Build] fota.download.http.socket.stm32wifi
[INFO-Build] fota.download.http.socket.stand
[INFO-Build] fota.download.coap
[INFO-Build] fsyscall
[INFO-Build] ywss
[INFO-Build] bone_engine
[INFO-Build] bluetooth.ble_app_framework
[INFO-Build] bluetooth.ais.ali_lib
[INFO-Build] bluetooth.ais.ble_app_ali
[INFO-Build] bluetooth.profile
[INFO-Build] gateway
[INFO-Build] gateway.devmgr
[INFO-Build] gateway.msdp
[INFO-Build] linkvoice
[INFO-Build] atparser
[INFO-Build] connectivity.mqtt
[INFO-Build] connectivity.mqtt.MQTTPacket
[INFO-Build] connectivity.coap
[INFO-Build] connectivity.websockets
[INFO-Build] connectivity.websockets.mbedtls_wrapper
[INFO-Build] connectivity.wsf
[INFO-Build] common
[INFO-Build] uData
[INFO-Build] libkm
[INFO-Build] tfs
[INFO-Build] plat_gen
[INFO-Build] alicrypto
[INFO-Build] mbedtls
[INFO-Build] prov
[INFO-Build] libid2
[INFO-Build] Lora
[INFO-Build] bluetooth.nrf51822
[INFO-Build] bluetooth.mk3239.ble_app_framework_impl
[INFO-Build] bluetooth.mk3239.bt_smart
[INFO-Build] bluetooth.mk3239.low_energy
[INFO-Build] bluetooth.mk3239.Firmware
[INFO-Build] bluetooth.mk3239.dual_mode
[INFO-Build] bluetooth.mk3239.ble_access
[INFO-Build] sal
[INFO-Build] sal.wifi.gt202
[INFO-Build] sal.wifi.mk3060
[INFO-Build] sensor
[INFO-Build] at_adapter
[INFO-Build] cli
[INFO-Build] base64
[INFO-Build] libc
[INFO-Build] hashtable
[INFO-Build] digest_algorithm
[INFO-Build] iotx-utils.sdk-impl
[INFO-Build] iotx-utils.mbedtls-hal
[INFO-Build] iotx-utils.digest
[INFO-Build] iotx-utils.iotx-system
[INFO-Build] iotx-utils.misc
[INFO-Build] iotx-utils.LITE-utils
[INFO-Build] iotx-utils.guider
[INFO-Build] iotx-utils.LITE-log
[INFO-Build] cJSON
[INFO-Build] log
[INFO-Build] hobbit1_evb
[INFO-Build] amebaz_dev
[INFO-Build] b_l475e
[INFO-Build] esp8266
[INFO-Build] eml3047
[INFO-Build] mk3166
[INFO-Build] esp32devkitc
[INFO-Build] linuxhost
[INFO-Build] lpcxpresso54102
[INFO-Build] frdmkl27z
[INFO-Build] xr871evb
[INFO-Build] mk3239
[INFO-Build] mk3060
[INFO-Build] mk1101
[INFO-Build] Below is a list of valid components from the internet:
[INFO-Build] 
[INFO-Build] 
[ERROR-Build] make.exe: *** No rule to make target 'c:/Users/cys/hello/out/helloworld@esp32devkitc/config.mk', needed by 'main_app'.  Stop.
[ERROR-Build] 
[ERROR-Build] [aos] ERROR: "c" returned error code 2.
[ERROR-Build] [aos] ERROR: Command "c : / U s e r s / c y s / h e l l o / a o s / b u i l d / c m d / w i n 3 2 / m a k e . e x e   - e   H O S T _ O S = W i n 3 2   T O O L S _ R O O T = c : / U s e r s / c y s / h e l l o / a o s / b u i l d   - f   b u i l d / M a k e f i l e   - e   h e l l o w o r l d @ e s p 3 2 d e v k i t c   A P P D I R = c : / U s e r s / c y s / h e l l o" in "c:\Users\cys\hello\aos"
[ERROR-Build] ---
[ERROR-Build] 
[ERROR-Build] -------------COMPILE OUTPUT FINISHED-------------
[ERROR-Build] Build Fail!
[INFO-Build] Compile time: 5.65s

librae8226, 2018-02-24:
请把您创建项目的步骤和编译命令也贴出来一下吧
文档请参考:
https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-Studio

librae8226, 2018-02-24:
aos make clean 或者删掉 out 目录应该就正常了，先close了哈
还有问题请reopen该issue

closed 2018-02-24 by librae8226