alibaba/AliOS-Things #147: 获取wifi信号强度的

LShaoPeng, 2018-03-02:
使用如下接口：

platform_wifi_get_rssi_dbm();

bk_wlan_get_link_status(&stat)；

获取到的wifi信号强度都为0，想了解一下，正确获取的方法，谢谢。

librae8226, 2018-03-02:
参考：
https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-API-HAL-WiFi-Guide#8-hal_wifi_get_link_stat
## 8-hal_wifi_get_link_stat

```c
int hal_wifi_get_link_stat(hal_wifi_module_t *m, hal_wifi_link_stat_t *out_stat)
```

- [x] **Description**

  Get the link status of the wifi instance ,e.g. ssid, bssid, channel, rssi, etc.

- [x] **Parameters**

  | IN/OUT |  NAME  |  DESC  |
  |--------|--------|--------|
  | [in] |  m |        the wifi instance, NULL if default.   |
  | [out] | out_stat | the place to hold the results. |

- [x] **Returns**

      0 on success, otherwise failure.

LShaoPeng, 2018-03-02:
所提供的函数之前也有使用过，传入默认参数  然后获取到的数值依旧是0

LShaoPeng, 2018-03-02:
使用方法如下：
hal_wifi_link_stat_t stat;
struct  hal_wifi_module_s *module_TEST   = hal_wifi_get_default_module();
ret = hal_wifi_get_link_stat(module_TEST, &stat);
打印信号强度值，一直为零

librae8226, 2018-03-05:
请提供一下您使用的硬件平台信息 @LShaoPeng

LShaoPeng, 2018-03-06:
EMW3060   EMW110

LShaoPeng, 2018-03-06:
@librae8226

librae8226, 2018-05-24:
目前暂时没有应用层获取wifi信号强度的统一接口。
对mk3060来说，这里面会引用到rssi:
  ./platform/mcu/moc108/hal/mesh_wifi_hal.c
不过这一处目前只有mesh在用

closed 2022-03-01 by skylarCai