alibaba/AliOS-Things #183: Aos Compile error

tonyho, 2018-04-02:
The aos command/script is not working. The environment is:
Ubuntu 16.04 X64
The code revision is at sh1a256: ef857aa06abc0958e9136e52580eb58b25efc311

The aos-cube is installed through pip. The Eroor log is:

```
$ aos make 
Traceback (most recent call last):
  File "/usr/local/bin/aos", line 11, in <module>
    load_entry_point('aos-cube==0.2.22', 'console_scripts', 'aos')()
  File "/usr/local/lib/python3.5/dist-packages/pkg_resources/__init__.py", line 572, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python3.5/dist-packages/pkg_resources/__init__.py", line 2755, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python3.5/dist-packages/pkg_resources/__init__.py", line 2408, in load
    return self.resolve()
  File "/usr/local/lib/python3.5/dist-packages/pkg_resources/__init__.py", line 2414, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.5/dist-packages/aos/aos.py", line 25, in <module>
    from .repo import *
  File "/usr/local/lib/python3.5/dist-packages/aos/repo.py", line 572
    sorted_scms = filter(lambda (m, _): m, sorted_scms)
                                ^
SyntaxError: invalid syntax

```

Cheng-SG, 2018-04-02:
Hi tonyho, Sorry for this inconvenience. Currently, aos-cube only supports Python 2.7. Please try using python 2.7 instead of Python 3.5. 

We will add Python 3 support  in the near future.

librae8226, 2018-04-08:
Close it for now due to comment above.
Please feel free to reopen it in case of any further problem.

closed 2018-04-08 by librae8226

zhf93139, 2019-11-12:
linux@linux:/$ aos make
Traceback (most recent call last):
  File "/usr/local/bin/aos", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2749, in <module>
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 446, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 459, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 632, in resolve
    raise VersionConflict(dist,req) # XXX put more info here
pkg_resources.VersionConflict: (pyserial 2.6 (/usr/lib/python2.7/dist-packages), Requirement.parse('pyserial>=3.0'))