alibaba/AliOS-Things #191: (Windows) Alios-studio:  新建项目问题，无项目选择页面。

zhuoshuguo, 2018-04-09:
@librae8226 

发现在当前的Visual studio code下（貌似经过一些相关的更新后），新建项目时出现问题，不再弹出项目[选择页面](https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-Studio#4-%E6%96%B0%E5%BB%BA%E4%B8%8E%E5%AF%BC%E5%85%A5%E9%A1%B9%E7%9B%AE)。
即不再“进入新建项目界面，选择创建应用对应的模板”。
以前是有选择页面的。

librae8226, 2018-04-09:
抱歉因为文档未及时更新给你造成的困扰。

为了支撑更大的定制化项目开发，我们正在迭代开发新建项目功能，目前 vs code 中创建项目的行为更新成与 `aos new [project name]` 命令一致，会在 vs code 窗口上方的命令栏弹出新工程名（可修改），然后为用户创建工程目录。
如果仍希望使用工程模版，可以直接拷贝 `example` 目录下的代码来使用。

另外，如果只是希望对示例代码（模板）进行编译、运行或简单修改，目前建议直接打开 AliOS Things 代码目录进行操作（不需要新建项目）。

zhuoshuguo, 2018-04-10:
> 为了支撑更大的定制化项目开发，我们正在迭代开发新建项目功能，目前 vs code 中创建项目的行为更新成与 aos new [project name] 命令一致，会在 vs code 窗口上方的命令栏弹出新工程名（可修改），然后为用户创建工程目录。如果仍希望使用工程模版，可以直接拷贝 example 目录下的代码来使用。

可以说的更仔细点么？我点击完"Create project"按钮之后，貌似新建了一个空的项目文件夹，
myProject/ , 项目文件夹里面只有：
```
.dist/
.vscode/
aos/  里面是空的
remote/
.aos
cube.mk
```

然后呢？把整个Alios的源码（AliOS-Things-master）都考到新建文件夹的aos/ 目录下？还是只考一部分？

比如我想用个helloworld模板，是只把example/helloworld/ 文件夹考到新建的myProject/aos/录下？

librae8226, 2018-04-11:
@zhuoshuguo 请尝试升级 aos-cube，然后删除用户目录下的 .aos 目录，再次 create project

closed 2022-03-01 by skylarCai