alibaba/AliOS-Things #196: AliOS Things Starter Kit哪里可以购买到？

fjyihan, 2018-04-12:
如题。以下链接中的AliOS Things Starter Kit开发板哪里可以买到？
https://github.com/aliosthings/aliosthings.github.io/blob/master/zh-cn/starterkit.md

librae8226, 2018-04-16:
很抱歉，目前仍然不能买到，预计要5月才会上线网店。我们会保持这个话题处于开放状态，直到可以买到了，会第一时间在此回复。
感谢关注！

ianhom, 2018-04-16:
你好，请问网点网址是什么

lzpfmh, 2018-05-07:
明天就要5月8号了，等待开放购买！

yuguitao, 2018-05-17:
可以购买了吗？

lzpfmh, 2018-05-23:
AliOS Things Developer Kit 又推出了这个，但是还没有购买链接，期待ing...

librae8226, 2018-05-28:
最新进展，6月5日会把网店上线，开放 Starter Kit 和 Developer Kit 购买。

ianhom, 2018-06-06:
Where are the Starter Kit & Developer Kit,please

lzpfmh, 2018-06-06:
6月6号了已经

siberianfox, 2018-06-06:
Starter Kit AliOS Things ：
https://detail.tmall.com/item.htm?spm=a220o.1000855.w4023-18510832497.4.bd4960deLuYxTn&id=571049481621

Developer Kit: AliOS Things
https://detail.tmall.com/item.htm?spm=a230r.1.14.1.482619e03J77iJ&id=571251003265&cm_id=140105335569ed55e27b&abbucket=13

已经上线了好像。

librae8226, 2018-06-08:
@siberianfox 是的。

https://opentogether.tmall.com/
请移步我们合作伙伴天猫店铺进行购买。
谢谢关注。

closed 2022-03-01 by skylarCai