alibaba/AliOS-Things #262: 基于esp32的linkkit example异常

zhangjinliang1985, 2018-05-16:
用esp32devkitc跑linkkit example会异常，用esp8266没有问题
---------------
submit an item to queue @0x3ffbeb88.
---------------
Guru Meditation Error of type StoreProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4000c46c  PS      : 0x00060a30  A0      : 0x8010596f  A1      : 0x3fff9560  
A2      : 0x00000000  A3      : 0x42424242  A4      : 0x3f40a40f  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x03f40a40  A8      : 0x80105f2a  A9      : 0x3fff95d0  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x3f40a7e2  A13     : 0x00000004  
A14     : 0x00000000  A15     : 0xff000000  SAR     : 0x00000007  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x03f40a3f  

Backtrace: 0x4000c46c:0x3fff9560 0x4010596c:0x3fff9570 0x40112da1:0x3fff9590

junjiec, 2018-05-16:
请问是master分支还是rel_1.3.0分支

zhangjinliang1985, 2018-05-16:
master 分支

junjiec, 2018-05-16:
@zhangjinliang1985 可以用rel_1.3.0试试吗

zhangjinliang1985, 2018-05-16:
还是有

zhangjinliang1985, 2018-05-16:
会出其他问题
handle_event_cb 7
[004810]<V> wifi_service_event config.ssid Noitom-RO
Guru Medets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
lo

closed 2022-03-01 by skylarCai