alibaba/AliOS-Things #283: linkkitapp ssl连接失败

maggie17, 2018-05-23:
在我们自己的硬件平台上移植了Alios，包括wifi hal和sal，运行linkkitapp时发现，每次在ssl连接过程都会出错，错误表现为以下几点：
1. 串口打印输出如下：可以看到在mbedtls_ssl_write_finished的时候socket意外关闭。

```
[inf] _TLSConnectNetwork(362):   . Setting up the SSL/TLS structure...
[inf] _TLSConnectNetwork(372):  ok
[inf] _TLSConnectNetwork(407): Performing the SSL/TLS handshake...
mbedtls_ssl_handshake_client_step: ssl->state = 0
mbedtls_ssl_handshake_client_step: ssl->state = 1
mbedtls_ssl_handshake_client_step: ssl->state = 2
mbedtls_ssl_handshake_client_step: ssl->state = 3
mbedtls_ssl_handshake_client_step: ssl->state = 4
mbedtls_ssl_handshake_client_step: ssl->state = 5
mbedtls_ssl_handshake_client_step: ssl->state = 6
mbedtls_ssl_handshake_client_step: ssl->state = 7
mbedtls_ssl_handshake_client_step: ssl->state = 8
mbedtls_ssl_handshake_client_step: ssl->state = 9
mbedtls_ssl_handshake_client_step: ssl->state = 10
mbedtls_ssl_handshake_client_step: ssl->state = 11
SOCK_CLOSE: index 0 handle 4350332 res 0
recv failed!
mbedtls_ssl_handshake_client_step: ssl->state = 12
```

2. 空口抓包发现，设备在发送Change Cipher Spec之后，马上会收到server发过来的TCP RST信息，导致socket关闭。
3. 之前使用alinkapp的时候没有发现该问题，即ssl能够成功完成对接。对比两者（alinkapp和linkkitapp）ssl连接过程中发送的数据，也没有发现什么异常，但是linkkitapp每次都会收到TCP RST信息，导致最后ssl失败。
4. alinkapp和linkkitapp使用的server是不一样的，linkkitapp的server在ssl连接过程中有什么不一样的吗？代码里是否需要什么特殊配置？

谢谢！

Ray148289, 2018-05-23:
是否有在iot_import_product.h中修改三元组呢

maggie17, 2018-05-23:
iot_import_product.h中改了以下参数：
PRODUCT_KEY
PRODUCT_SECRET
DEVICE_NAME
DEVICE_SECRET
依然会有上述问题出现

maggie17, 2018-05-28:
附上完整的串口打印信息如下：

```
shell> Connected to Hydra, Auth completed

shell> _dhc_setip Got IP address 10.75.58.181
DNS 0: 10.253.157.26
DNS 1: 10.253.157.27
[012300]<V> wifi_service_event config.ssid Hydra
linkkit_main@line470:   sample execution time: 1 minutes
linkkit_main@line471:   Not get tsl from cloud
log_cls constructed @0x48ac38
slist_cls constructed @0x48ac58
slist_cls constructed @0x48ac98
slist_cls constructed @0x48acd8
slist_cls constructed @0x48ad18
slist_cls constructed @0x48ad58
slist_cls constructed @0x48ad98
cm_msg_info_cls constructed @0x494fd0
dm_cm_impl_cls constructed @0x48add8
slist(callback list) insert node,size:1
[inf] iotx_device_info_init(40): device_info created successfully!
[dbg] iotx_device_info_set(50): start to set device info!
[dbg] iotx_device_info_set(64): device_info set successfully!
[dbg] guider_print_dev_guider_info(248): ....................................................
[dbg] guider_print_dev_guider_info(249):           ProductKey : b1OR5pgr0l4
[dbg] guider_print_dev_guider_info(250):           DeviceName : u1bpUt9rkbk0m9xfKtFj
[dbg] guider_print_dev_guider_info(251):             DeviceID : b1OR5pgr0l4.u1bpUt9rkbk0m9xfKtFj
[dbg] guider_print_dev_guider_info(253): ....................................................
[dbg] guider_print_dev_guider_info(254):        PartnerID Buf : ,partner_id=AliOSThings
[dbg] guider_print_dev_guider_info(255):         ModuleID Buf : ,module_id=GeneralID
[dbg] guider_print_dev_guider_info(256):           Guider URL : 
[dbg] guider_print_dev_guider_info(258):       Guider SecMode : 2 (TLS + Direct)
[dbg] guider_print_dev_guider_info(260):     Guider Timestamp : 2524608000000
[dbg] guider_print_dev_guider_info(261): ....................................................
[dbg] guider_print_dev_guider_info(267): ....................................................
[dbg] guider_print_conn_info(225): -----------------------------------------
[dbg] guider_print_conn_info(226):             Host : b1OR5pgr0l4.iot-as-mqtt.cn-shanghai.aliyuncs.com
[dbg] guider_print_conn_info(227):             Port : 1883
[dbg] guider_print_conn_info(230):         ClientID : b1OR5pgr0l4.u1bpUt9rkbk0m9xfKtFj|securemode=2,timestamp=2524608000000,signmethod=hmacsha1,gw=0,partner_id=AliOSThings,module_id=GeneralID|
[dbg] guider_print_conn_info(232):       TLS PubKey : 0x450edc ('-----BEGIN CERTIFICATE-----
MIIDdTCCAl2gAwIBAgILBAAAAAABFUtaw5QwDQYJKoZIhvcNAQEFBQAwVzELMAkG
A1UEBhMCQkUxGTAXBgNVBAoTEEdsb2JhbFNpZ24gbnYtc2ExEDAOBgNVBAsTB1Jv
b3QgQ0ExGzAZBgNVBAMTEkdsb2JhbFNpZ24gUm9vdCBDQTAeFw05ODA5MDExMjAw8¬H ...
[dbg] guider_print_conn_info(235): -----------------------------------------
[inf] IOT_CM_Init(117): cm context initialize
linked_list_insert@line101      :linked list(cm event cb list) insert node@0x48af18,size:1
linked_list_insert@line101      :linked list(cm connectivity list) insert node@0x48af38,size:1
[inf] iotx_mc_init(1705): MQTT init success!
[inf] _ssl_client_init(176): Loading the CA root certificate ...
cert. version     : 3
serial number     : 04:00:00:00:00:01:15:4B:5A:C3:94
issued  on        : 0000-00-00 00:00:00
expires on        : 0000-00-00 00:00:00
RSA key size      : 2048 bits
[inf] _ssl_parse_crt(144): crt content:185
[inf] _ssl_client_init(184):  ok (0 skipped)
[inf] _TLSConnectNetwork(345): Connecting to /b1OR5pgr0l4.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...

Looked up b1OR5pgr0l4.iot-as-mqtt.cn-shanghai.aliyuncs.com as 139.196.135.135
SOCK_OPEN: index 0 handle 4350332
SOCK_CONNECT: index 0 handle 4350332 res 0
[inf] _TLSConnectNetwork(362):   . Setting up the SSL/TLS structure...
[inf] _TLSConnectNetwork(372):  ok
[inf] _TLSConnectNetwork(407): Performing the SSL/TLS handshake...
mbedtls_ssl_handshake_client_step: ssl->state = 0
mbedtls_ssl_handshake_client_step: ssl->state = 1
mbedtls_ssl_handshake_client_step: ssl->state = 2
mbedtls_ssl_handshake_client_step: ssl->state = 3
mbedtls_ssl_handshake_client_step: ssl->state = 4
mbedtls_ssl_handshake_client_step: ssl->state = 5
mbedtls_ssl_handshake_client_step: ssl->state = 6
mbedtls_ssl_handshake_client_step: ssl->state = 7
mbedtls_ssl_handshake_client_step: ssl->state = 8
mbedtls_ssl_handshake_client_step: ssl->state = 9
mbedtls_ssl_handshake_client_step: ssl->state = 10
mbedtls_ssl_handshake_client_step: ssl->state = 11
SOCK_CLOSE: index 0 handle 4350332 res 0
recv failed!
mbedtls_ssl_handshake_client_step: ssl->state = 12
```

closed 2018-05-29 by maggie17

junjiec, 2018-05-29:
@maggie17 问题解决了？可以写下解决办法吗？

maggie17, 2018-05-29:
@junjiec Hi，应该是网络问题，之前一直用的公司wifi，ssl连接出错，今天换了个别的wifi网络（手机热点），就能正常连接了