alibaba/AliOS-Things #315: 咨询AliOS Thing系统对RAM、Flash存储的资源占用分别是多少？谢谢

lijk6455352, 2018-06-01:
在AliOS Thing的基础上，配置上CoAP、MQTT 等常用协议， 总共会占用多少的RAM？ flash存储总共占用多少？
谢谢。

lijk6455352, 2018-06-01:
Ask for a question about hao many RAM & ROM will be used running AliOS Thing OS with CoAP & MQTT .

librae8226, 2018-06-08:
根据实际的硬件和编译的应用决定，可以尝试编译一个目标，会在日志中生成 AOS MEMORY MAP

比如可以参考 `aos make mqttapp@mk3060` 的结果。
这里面也可以根据需要去配置和裁剪，以契合最终的产品需要。
```
                        AOS MEMORY MAP
|=================================================================|
| MODULE                                   | ROM       | RAM      |
|=================================================================|
| newlib_stub                              | 665       | 0        |
| alicrypto                                | 40082     | 8780     |
| kv                                       | 2993      | 36       |
| fota                                     | 2989      | 345      |
| socket_stand                             | 1075      | 0        |
| libc_nano                                | 28436     | 464      |
| vcall                                    | 6756      | 4        |
| kernel_init                              | 1141      | 48       |
| yloop                                    | 1555      | 24       |
| rhino                                    | 14091     | 7721     |
| libplat_gen                              | 469       | 0        |
| framework                                | 317       | 12       |
| vfs                                      | 1821      | 1209     |
| digest_algorithm                         | 2538      | 0        |
| libgcc                                   | 6136      | 0        |
| fota_download                            | 211       | 0        |
| activation                               | 2743      | 8776     |
| fota_mqtt_transport                      | 4278      | 58       |
| libprov                                  | 596       | 0        |
| download_http                            | 2046      | 88       |
| cli                                      | 6033      | 549      |
| mqtt                                     | 14519     | 20       |
| board_mk3060                             | 649       | 24       |
| fota_platform                            | 76        | 0        |
| moc108                                   | 165473    | 52793    |
| armv5                                    | 388       | 0        |
| netmgr                                   | 1887      | 266      |
| mqttapp                                  | 1371      | 268      |
| mbedtls                                  | 28230     | 76       |
| cjson                                    | 2009      | 20       |
| net                                      | 46476     | 3676     |
| iotkit                                   | 15893     | 1705     |
| log                                      | 414       | 20       |
| MQTTPacket                               | 2063      | 4        |
| hal                                      | 298       | 12       |
| *fill*                                   | 466       | 142      |
|=================================================================|
| TOTAL (bytes)                            | 407183    | 87140    |
|=================================================================|
```

closed 2022-03-01 by skylarCai