alibaba/AliOS-Things #326: aos-cube 依赖 scons 一直安装不成功

cxlwill, 2018-06-04:
问题描述： 环境下，`aos-cube` 依赖 `scons` 一直安装不成功，导致 `aos-cube` 无法安装，在实验用机均出现此状况。

环境：OS X 10.13，Python 2.7.15, `setuptools` 39.2.0, `wheel` 0.31.1

错误代码：

```shell
Running setup.py install for scons ... error

    Complete output from command /Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u -c "import setuptools, tokenize;__file__='/private/tmp/pip-install-RIQgaT/scons/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/tmp/pip-record-pxsHqb/install-record.txt --single-version-externally-managed --compile --user --prefix=:

    usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]

       or: -c --help [cmd1 cmd2 ...]

       or: -c --help-commands

       or: -c cmd --help
  error: option --single-version-externally-managed not recognized
    ----------------------------------------

Command "/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u -c "import setuptools, tokenize;__file__='/private/tmp/pip-install-RIQgaT/scons/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/tmp/pip-record-pxsHqb/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /private/tmp/pip-install-RIQgaT/scons/
```

librae8226, 2018-06-04:
谢谢，很多朋友遇到了
请更新下 python 的 `setuptools` 和 `wheel` 包
```
pip install --upgrade setuptools
pip install --upgrade wheel
```

cxlwill, 2018-06-04:
已经是最新版了，还是出错。

librae8226, 2018-06-05:
这样的话推荐使用 `virtualenv`
```
pip install virtualenv
virtualenv ~/.virtualenvs/python-venv
. ~/.virtualenvs/python-venv/bin/activate
pip install --upgrade setuptools
pip install --upgrade wheel
pip install aos-cube
```

wgj1922, 2018-06-05:
我安装时也碰到了这个问题，按照网上给的各种方法试过，都解决不了。
后来我是把安装包下载到本地安装成功的，方法如下，可以参考：
* 先下载你要安装的包，并解压到磁盘下（待下载链接在执行pip install命令失败时，会打印出来，将这个链接拷贝出来，在浏览器中输入就可以直接下载本地安装包）；
* 命令行方式，进入到该安装包的setup.py 目录下 ；
* 执行 python setup.py build
* 执行 python setup.py install

cxlwill, 2018-06-11:
@wgj1922 亲测可行，非常感谢！

laoyouji007, 2018-06-19:
@wgj1922 
解压安装完scons之后，还要再次运行一下命令才行：pip install --user aos-cube

dupig, 2018-08-10:
在Mac上调用路径不正确会导致烧录时scons无效。解决方法是去VS终端输入which scons看看是不是在python 2.7目录下。如果是在系统的usr/local/bin目录下则在finder里删掉，然后调用路径就会更新回python 2.7的目录就OK了。

closed 2022-03-01 by skylarCai