alibaba/AliOS-Things #337: 建议JS的部分参考github上的Espruino 项目

walker2048, 2018-06-07:
Espruino 是个可以下载预编译好的bin文件烧录到机子上，直接在IDE写JS文件就能在线运行的。也就是说使用者甚至不需要知道怎样编译，直接编写js程序就行。

对于业余编程人员和部分注重快速开发人员来说，目前项目的复杂度有点大了吧。
例如STM32L476的板子，用aos-cube新建好的例子项目都没法顺利编译。
毕业了就没碰过C语言了，编译和makefile什么的真心有点吃力。
既然考虑用JS框架，是否考虑下这种实现方式？

librae8226, 2018-06-08:
了解的，目前刚刚发布 tinyengine，后续还有很多更新，请继续提出宝贵意见。

wuzhun1001, 2018-06-11:
我们内部已有一个类似的Project -- Gravity ，是基于BoneEngine的，专门面向 JS Apps 应用开发者的，官方会提供预编译的 BoneEngine 系统固件，并已经支持了各种芯片和外设，同时它将带有 be-cli 和 WebIDE 等JS 开发调试工具，这个项目的目标是打造BoneEngine的JS应用和设备生态，将会很快开源出来。

walker2048, 2018-06-11:
想问下这个项目的硬件要求如何，例如多大的FLASH，多大的RAM

wuzhun1001, 2018-06-14:
tinyEngine全功能插件支持的版本，ROM占用为70K Bytes，RAM占用15K Bytes。对于资源紧张的场景，可以通过配置native addon插件模块支持情况减少资源的占用，最小精简版的tinyEngine ROM<15K ,RAM<10K 。
同时tinyEngine默认划分了一块FLASH分区为SPIFFS文件系统分区，这块分区可以根据实际情况自行调节。

boneq, 2018-07-28:
tinyEngine 中的js生成的bin文件除了be push 还有其它方式上传到设备么？

hanfengcan, 2018-11-21:
考虑像nodemcu可以选择模块编译，减少体积

closed 2022-03-01 by skylarCai