alibaba/AliOS-Things #356: AliOS Things Studio 无法烧录ESP32开发板

jyxk, 2018-06-17:
```
> Executing task: aos upload helloworld@esp32devkitc <

[INFO]:target: helloworld@esp32devkitc
[INFO]:Currently in aos_sdk_path: 'D:\AliOS-Things-master'

--- Available ports:
---  1: COM3                 u'Silicon Labs CP210x USB to UART Bridge (COM3)'
--- Enter port index or full name: 1

[INFO]:image_path: D:/AliOS-Things-master/out/helloworld@esp32devkitc/binary/helloworld@esp32devkitc.bin
[INFO]:bootloader_path: D:/AliOS-Things-master/platform/mcu/esp32/bsp/bootloader.bin
[INFO]:partition_path: D:/AliOS-Things-master/platform/mcu/esp32/bsp/custom_partitions.bin

[INFO]:exec_cmd: ['esptool.py', '--chip', 'esp32', '--port', u'COM3', '--baud', '921600', '--before', 'default_reset', '--after', 'hard_reset', 'write_flash', '-z', '--flash_mode', 'dio', '--flash_freq', '40m', '--flash_size', 'detect', '0x1000', 'D:/AliOS-Things-master/platform/mcu/esp32/bsp/bootloader.bin', '0x8000', 'D:/AliOS-Things-master/platform/mcu/esp32/bsp/custom_partitions.bin', '0x10000', 'D:/AliOS-Things-master/out/helloworld@esp32devkitc/binary/helloworld@esp32devkitc.bin']
'esptool.py' 不是内部或外部命令，也不是可运行的程序
或批处理文件。
---host_os:Win32
firmware upload failed

终端将被任务重用，按任意键关闭。
```

Faynman, 2018-10-26:
升级后还是不能解决 @librae8226

librae8226, 2018-10-26:
> 升级后还是不能解决 @librae8226 
不能解决的话问题是啥，贴个日志？

kunta87, 2020-09-18:
---  2: COM4                 'USB-SERIAL CH340 (COM4)'
--- Enter port index or full name: 2
[INFO]: Running cmd:
        'esptool.exe --chip esp8266 --port COM4 --baud 921600 write_flash --flash_size detect 0x0 C:\Users\Administrator\AliOS-Things/platform/mcu/esp8266/bsp/boot_v1.7_921600.bin 0x3fc000 C:\Users\Administrator\AliOS-Things/platform/mcu/esp8266/bsp/esp_init_data_default.bin 0x3fe000 C:\Users\Administrator\AliOS-Things/platform/mcu/esp8266/bsp/blank.bin 0x1000 C:\Users\Administrator\AliOS-Things/out/blink_demo@esp8266/binary/blink_demo@esp8266.bin'    
'esptool.exe' 不是内部或外部命令，也不是可运行的程序
或批处理文件。
---host_os:Win32
[ERROR]: Firmware upload failed!

终端进程“C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command aos upload blink_demo@esp8266”已终止，退出代码: 1。

终端将被任务重用，按任意键关闭。

closed 2022-03-01 by skylarCai