alibaba/AliOS-Things #360: CLI 串口无法打印

LeoZhang-ATMEL, 2018-06-20:
同步[最新代码](https://github.com/alibaba/AliOS-Things/commit/6177c74febb7c4ffd030a7de79f9c37fcc47b832)后，CLI无法打印，uart_stdio.port被强制设置位0，导致串口无法打印。
https://github.com/alibaba/AliOS-Things/blob/6177c74febb7c4ffd030a7de79f9c37fcc47b832/tools/cli/cli.c#L782-L812

LeoZhang-ATMEL, 2018-06-21:
由于hal中会用到uart_dev_t->priv 来找到对应的私有结构体，是否也要在hal中指定？
程序的维护会变得比较麻烦，而且也没有相应的文档。

jason-sophia, 2018-06-22:
是的，这种情况下priv也会不可用，标准输入输出是个特例，你可以对其进行特殊处理，其他的设备都是可以使用的，我们会在文档中对这个进行说明。

NashZhangMCHP, 2018-08-09:
在2018.8.7的版本里，看到这样的代码：
```
// AliOS-Things/tools/cli/cli.c
int cli_putstr(char *msg)
{
    uart_dev_t uart_stdio;

    memset(&uart_stdio, 0, sizeof(uart_stdio));
    uart_stdio.port = STDIO_UART;

    if (msg[0] != 0) {
        hal_uart_send(&uart_stdio, (void *)msg, strlen(msg), HAL_WAIT_FOREVER);
    }

    return 0;
}
```
函数开始把uart_stdio清零，然后把STDIO_UART赋值给port。如果hal中会用到uart_dev_t->priv，.priv每次都是0，代码执行会出现问题. 类似的代码在AliOS-Things/utility/libc/newlib_stub.c 中也有。

建议：设置一个全局变量uart_stdio且uart_stdio.port初值为STDIO_UART. 这样在函数里，就不需要uart_studio了。这个全局变量cli.c和newlib_stub.c共用。

jason-sophia, 2018-08-09:
@LeoZhang-ATMEL 你说的问题是存在的，但是不用priv也没有那么的麻烦，你只需要在hal_uart_send函数中根据port获取原本在hal_uart_init中获取到的priv指针即可，可以参考我们STM32L4系列的hal对接代码

jason-sophia, 2018-08-09:
@NashZhangMCHP  其实我们这样设计主要是为了兼容多bin，在多bin的情况下，app和kernel是分开的。标准输入输出是由用户进行初始化，也就是其对应的uart_dev_t结构体是定义在app中的，但是kernel中要进行异常输出和CLI交互等功能，也需要使用到该结构体，kernel无法获取app中的该结构体。由于kernel中只需要进行send和recive操作而不需要初始化，因此kernel中需要进行标准输入输出时，我们重新定义一个uart_dev_t结构体，将port设置为0即可。由于这个结构体不是初始化时app传入的结构体，因此这个结构体中除了port之外其他的成员都是不可用的。具体来说就是针对标准输入输出串口，在数据发送和接收函数中不要使用uart_dev_t结构体中的priv和config元素，因为这两个元素在kernel中是无效的。对于其他串口，还有其他设备，无此限制。

jason-sophia, 2018-08-09:
https://yq.aliyun.com/articles/624057
这篇文章是介绍具体如何对接和修改标准输入输出端口的，可以看看

NashZhangMCHP, 2018-08-10:
@jason-sophia 谢谢详细的解答，明白原因了。我会提交一个pull request，对hal uart进行修改，不然2018.8.7的版本是无法工作的。

closed 2022-03-01 by skylarCai