alibaba/AliOS-Things #373: esp8266运行linkkitapp后不能进行配网！

sofwu, 2018-06-24:
esp8266运行linkkitapp后不能进行配网！串口打印如下
[image]

librae8226, 2018-06-25:
有触法配网流程么？

sofwu, 2018-06-25:
@librae8226 esp8266的开发板不知道怎样触发配网

sofwu, 2018-06-25:
[image]
下面是demo中别的硬件触发配网的监听程序
void linkkit_key_process(input_event_t *eventinfo, void *priv_data)
{
    if (eventinfo->type != EV_KEY) {
        return;
    }
    LOG("awss config press %d\n", eventinfo->value);

    if (eventinfo->code == CODE_BOOT) {
        if (eventinfo->value == VALUE_KEY_CLICK) {
            do_awss_active();
        } else if (eventinfo->value == VALUE_KEY_LTCLICK) {
            do_awss_reset();
        }
    }
}

gofortime, 2018-06-25:
@sofwu 开发版的J70和J68的SCL引脚（下图用绿色圈起来的地方）用跳线连起来，再通过SW2按键进入配网。
[image]

sofwu, 2018-06-25:
配网成功之后又自动重启了，app上点击开始使用后一直处于等待中最后提示绑定失败[image]
[image]

sofwu, 2018-06-25:
@gofortime 配网成功之后自动重启了这个问题能帮忙看看吗

judsonzhu, 2018-07-23:
@sofwu 你好，你是用demo APP配网吗，我用8266 AliOS 1.3.2一直配网不上，WiFi一直在切换信道，接收不到配网数据呢

closed 2018-07-30 by sofwu

mq9494, 2019-01-25:
我现在也处于配网成功，绑定失败状态。不知道你的问题解决没，请教一下。同样用的ESP8266