alibaba/AliOS-Things #375: Example的Blink@stm32f429zi-nucleo 編譯失敗

r1244460xx, 2018-06-25:
Environmet: Windows AliOS Studio
Helloworld@stm32f429zi-nucleo、Blink@starterkit編譯會成功
但Blink@@stm32f429zi-nucleo編譯失敗
LOG：
PS C:\Users\Severus\AliOS-Things> aos make blink@stm32f429zi-nucleo
Making config file for first time
processing components: blink stm32f429zi-nucleo platform/mcu/stm32f4xx_cube vcall init auto_component
kernel_version:AOS-R-1.3.1
kernel_version:AOS-R-1.3.1
Build AOS Now
TOOLCHAIN_PATH=C:/Users/Severus/AliOS-Things/build/compiler/gcc-arm-none-eabi/Win32/bin/
Compiling auto_component
Compiling blink
Compiling cli
Compiling digest_algorithm
Compiling hal
Compiling kernel_init
Compiling vfs_device
Compiling newlib_stub
Compiling log
Compiling armv7m
Compiling stm32f4xx_cube
Compiling net
Compiling rhino
Compiling stm32f429zi-nucleo
Compiling vcall
Compiling vfs
Compiling yloop
Making out/blink@stm32f429zi-nucleo/libraries/auto_component.a
Making out/blink@stm32f429zi-nucleo/libraries/blink.a
Making out/blink@stm32f429zi-nucleo/libraries/cli.a
Making out/blink@stm32f429zi-nucleo/libraries/digest_algorithm.a
Making out/blink@stm32f429zi-nucleo/libraries/hal.a
Making out/blink@stm32f429zi-nucleo/libraries/kernel_init.a
Making out/blink@stm32f429zi-nucleo/libraries/vfs_device.a
Making out/blink@stm32f429zi-nucleo/libraries/newlib_stub.a
Making out/blink@stm32f429zi-nucleo/libraries/log.a
Making out/blink@stm32f429zi-nucleo/libraries/armv7m.a
Making out/blink@stm32f429zi-nucleo/libraries/stm32f4xx_cube.a
Making out/blink@stm32f429zi-nucleo/libraries/net.a
Making out/blink@stm32f429zi-nucleo/libraries/rhino.a
Making out/blink@stm32f429zi-nucleo/libraries/stm32f429zi-nucleo.a
Making out/blink@stm32f429zi-nucleo/libraries/vcall.a
Making out/blink@stm32f429zi-nucleo/libraries/vfs.a
Making out/blink@stm32f429zi-nucleo/libraries/yloop.a
Making blink@stm32f429zi-nucleo.elf
out/blink@stm32f429zi-nucleo/libraries/blink.a(blink.o): In function `app_trigger_low_action':
C:\Users\Severus\AliOS-Things/example/blink/blink.c:46: undefined reference to `hal_gpio_output_low'
out/blink@stm32f429zi-nucleo/libraries/blink.a(blink.o): In function `app_trigger_high_action':
C:\Users\Severus\AliOS-Things/example/blink/blink.c:52: undefined reference to `hal_gpio_output_high'
out/blink@stm32f429zi-nucleo/libraries/blink.a(blink.o): In function `gpio_isr_handler':
C:\Users\Severus\AliOS-Things/example/blink/blink.c:39: undefined reference to `hal_gpio_input_get'
C:\Users\Severus\AliOS-Things/example/blink/blink.c:40: undefined reference to `hal_gpio_output_toggle'
out/blink@stm32f429zi-nucleo/libraries/blink.a(blink.o): In function `application_start':
C:\Users\Severus\AliOS-Things/example/blink/blink.c:63: undefined reference to `hal_gpio_init'
C:\Users\Severus\AliOS-Things/example/blink/blink.c:70: undefined reference to `hal_gpio_init'
C:\Users\Severus\AliOS-Things/example/blink/blink.c:77: undefined reference to `hal_gpio_init'
C:\Users\Severus\AliOS-Things/example/blink/blink.c:80: undefined reference to `hal_gpio_enable_irq'
collect2.exe: error: ld returned 1 exit status
make.exe[1]: *** [build/aos_target_build.mk:326: out/blink@stm32f429zi-nucleo/binary/blink@stm32f429zi-nucleo.elf] Error 1
make.exe: *** [build/Makefile:267: main_app] Error 2
[AliOS-Things] ERROR: "C" returned error code 2.
[AliOS-Things] ERROR: Command "C : / U s e r s / S e v e r u s / A l i O S - T h i n g s / b u i l d \ c m d / w i n 3 2 / m a k e . e x e
  H O S T _ O S = W i n 3 2   T O O L S _ R O O T = C : / U s e r s / S e v e r u s / A l i O S - T h i n g s / 

上面顯示找不到原始碼檔，我去trace stm32f4xx_cube的mk檔，發現hal_gpio被註解掉了
$(NAME)_SOURCES += aos/soc_impl.c \
                   aos/trace_impl.c \
                   aos/aos.c \
                   hal/hal_uart_stm32f4.c \
                   hal/hw.c \
                   hal/hal_flash_stm32f4.c #\
                   hal/hal_sd_stm32f4.c \
                   hal/hal_adc_stm32f4.c \
                   hal/hal_i2c_stm32f4.c \
                   hal/hal_gpio_stm32f4.c \
                   hal/hal_rtc_stm32f4.c \
                   hal/hal_spi_stm32f4.c \
                   hal/hal_qspi_stm32f4.c \
                   hal/hal_nand_stm32f4.c \
                   hal/hal_nor_stm32f4.c
將它解註解後會出現其他的問題，所以還是回報來這邊
麻煩大家一起幫忙了，有需要更詳細的錯誤資訊我再更新，謝謝~

walker2048, 2018-06-26:
现在阿里的资源估计都在starterkit和developerkit，还有esp8266？
现在官方库里，其他板子是有些问题的
一是板子资源定义（包括通讯端口数量，Pin脚问题）
二是板子hal实现库和stm32官方HAL库实现是否一致的问题
我是直接自己修改自己的板子资源，stm32l476rg-nucleo板子路过

librae8226, 2018-06-26:
目前 stm32f429zi-nucleo hal 部分的对接缺失一些东西，如果需要使用可以参考其他st系列板子把缺的部分加上。

closed 2022-03-01 by skylarCai