alibaba/AliOS-Things #399: linkkitapp示例中的设备激活凭证能不能不写死在程序内

nostalgia-w, 2018-07-04:
linkkitapp的示例中，需要根据云端产生的设备凭证，修改头文件framework/protocol/linkkit/iotkit/sdk-encap/imports/iot_import_product.h下的设备凭证信息。
#define PRODUCT_KEY               "a1BDCKKSpdu"
#define PRODUCT_SECRET          "Rp0aCxyRnd2ecrCX"
#define DEVICE_NAME                "MVwo1x7hB2pyEbXlLkzE"
#define DEVICE_SECRET              "OKHWuIaMzwtJAjUzPRLG1rcYxFo43akU"
但是假如设备量产，那肯定不可能每个设备都修改程序内的这个头文件，然后编译烧录。
而且固件升级的时候，假如这个设备凭证是写死在程序中的，那批量固件升级的时候就全变成一个设备了。
想问一下，有没有办法可以像wifi账号密码一样，不写死的程序内，而是通过别的方式，比如用串口通过kv set将这个激活凭证写在设备内？

nostalgia-w, 2018-07-09:
framework/protocol/linkkit/iotkit/hal-impl/rhino/HAL_PRODUCT_rhino.c，使用HAL_GetDeviceSecret函数就可以解决上面的问题，前提板子支持KV。
方法：可以先试试HAL_SetDeviceSecret函数，然后看板子里的kv list，就知道HAL_GetDeviceSecret函数怎么用了。

xuhongv, 2018-09-12:
请问 esp8266板子支持KV吗

nostalgia-w, 2018-09-13:
支持

发自我的 iPhone

nostalgia-w, 2018-09-13:
> 请问 esp8266板子支持KV吗

我用的是esp-wroom-02，内核是esp8266，是有KV的，就是你存的地址要自己改一下
可以先从github上的文档了解一下https://github.com/alibaba/AliOS-Things/wiki/Flash-and-KV-Storage-Porting-Guide.zh
 在cli下面1. 先kv list看下输出 有木有输出  2. kv set test abc看看能否设下去 3. kv get test看看读出来的值是不是设下去的值

closed 2022-03-01 by skylarCai