alibaba/AliOS-Things #400: linkkitapp@esp32devkitc 异常log且无法连接阿里云 linkkitapp@esp32developerkit上正常运行

leoadvance, 2018-07-05:
如题，上电后一直打印如下log，通过netmgr配网后不会有连接阿里云动作发生。请问是什么问题？

```
wifi_getset_ops 278: channe - 3
chan 3
wifi_getset_ops 278: channe - 6
chan 6
wifi_getset_ops 278: channe - 7
chan 7
wifi_getset_ops 278: channe - 11
chan 11
wifi_getset_ops 278: channe - 13
chan 13
wifi_getset_ops 278: channe - 1
chan 1
wifi_getset_ops 278: channe - 6
chan 6
wifi_getset_ops 278: channe - 11
chan 11
wifi_getset_ops 278: channe - 1
chan 1
wifi_getset_ops 278: channe - 2
chan 2
wifi_getset_ops 278: channe - 3
chan 3
wifi_getset_ops 278: channe - 4
chan 4
wifi_getset_ops 278: channe - 5
chan 5
wifi_getset_ops 278: channe - 6
chan 6
wifi_getset_ops 278: channe - 7
chan 7
```

librae8226, 2018-07-05:
看起来设备没有进入配网状态，cli中运行一下“active_awss”命令进入配网

leoadvance, 2018-07-05:
运行
active_awss
得到
do_awss_active 1
然后netmgr配网 并没有连接阿里云动作，之前log依旧存在。

librae8226, 2018-07-09:
@leoadvance 还有更完整的日志吗，从启动开始，到命令执行之后的

librae8226, 2018-07-09:
@gofortime 看看这个问题呢

gofortime, 2018-07-09:
@leoadvance 有配网软件发配网包吗？从你的log看，没有收到任何配网包。

leoadvance, 2018-07-10:
@gofortime 有用netmgr配网，能看到配网成功，但是依然有如上log，且不会连接“智能生活开发平台”。 而在developerkit上就能正常连接“智能生活开发平台”，你们是否有ESP32，直接跑下代码？

gofortime, 2018-07-11:
@leoadvance 你是通过netmgr connect联网？这个原来支持的，现在这个不支持了。你可以改一下代码来支持netmgr connect，具体修改可以参考附件的patch。
[image]

leoadvance, 2018-07-11:
@gofortime 十分感谢，按照提示修改后问题解决了。问题是。1.为什么developerkit未修改前就正常。2.以后不用netmgr配网那用什么配网？

lluwang, 2018-07-11:
目前节点启动后，直接进入monitor模式，是应产品的需要。
后续我们会增加功能，输入netmgr后，会退出monitor模式，等待用户输入netmgr connect功能。

closed 2018-07-12 by leoadvance