alibaba/AliOS-Things #412: yts的test_yloop.aos_2_004测试例子有误

elfplz, 2018-07-12:
**Describe the bug**
在yts的入口处，就是文件 example/yts/main.c里面的application_start中，会调用了yloop_run，这样，启动测试的任务就进入了yloop循环。
test/testcase/certificate_test/aos_test.c  的测试aos_2_004中，会在设置了event callback之后，马上就post了event了，这会导致对应的 ev_callback1马上就被调用，然后aos_loop_exit就调用了。这样最初的任务就被退出了。接下来aos_2_004在post之后会调用aos_loop_run，但是这个时候因为之前post的event已经被处理了，这样就会在这里死循环了。

建议不要在yts入口处调用yloop_run，而是换成忙等之类的，example/yts/main.c中的yloop_run替换成
while (1) {aos_msleep(1000); }

**environment**
1. branch(version): 
2. Desktop OS(windows linux or macOS):
3. example: 

**Expected behavior**

**Screenshots**

**Additional context**

Jianlong0430, 2018-07-12:
感谢反馈，我们内部同学在确认这个问题。

Jianlong0430, 2018-07-16:
Hi @elfplz ,
Please see comments as below and double check if this issue is fixed.

root cause：aos_post_event不与当前事件循环绑定，在事件回调中调用aos_loop_exit会退出主循环，进而阻塞用例中的循环
how to fix：事件回调中去掉aos_loop_exit，test_yloop.aos_2_004中通过delayed_action退出当前循环
side effects：无

diff --git a/test/testcase/certificate_test/aos_test.c b/test/testcase/certificate_test/aos_test.c
index d594897..761c134 100644
--- a/test/testcase/certificate_test/aos_test.c
+++ b/test/testcase/certificate_test/aos_test.c
@@ -907,7 +907,6 @@ static void ev_callback1(input_event_t *event, void *private_data)
 {
     printf("ev_callback1 is called\r\n");
     g_var = 0x5A;
-    aos_loop_exit();
 }
 
 static void delayed_action1(void *arg)
@@ -966,12 +965,15 @@ CASE(test_yloop, aos_2_004)
     ret = aos_register_event_filter(EV_USER_TYPE1, ev_callback1, NULL);
     ASSERT_EQ(ret, 0);
 
+    ret = aos_post_delayed_action(1000, delayed_action2, NULL);
+    ASSERT_EQ(ret, 0);
+
     ret = aos_post_event(EV_USER_TYPE1, 0, 0);
     // ASSERT_EQ(ret, 0);
 
     aos_loop_run();
     aos_unregister_event_filter(EV_USER_TYPE1, ev_callback1, NULL);
-    ASSERT_EQ(g_var, 0x5A);
+    ASSERT_EQ(g_var, 0x5B);
 }
 
 CASE(test_yloop, aos_2_005)
diff --git a/test/testcase/certificate_test/certificate_test.mk b/test/testcase/certificate_test/certificate_test.mk
index d6ed416..048151c 100644
--- a/test/testcase/certificate_test/certificate_test.mk
+++ b/test/testcase/certificate_test/certificate_test.mk
@@ -5,4 +5,4 @@ $(NAME)_COMPONENTS := rhino
 $(NAME)_COMPONENTS += modules.fs.kv
 $(NAME)_COMPONENTS += yloop
 
-$(NAME)_CFLAGS += -Wall -Werror -Wno-unused-varible
+$(NAME)_CFLAGS += -Wall -Werror -Wno-unused-variable

elfplz, 2018-07-16:
@Jianlong0430 我觉得这样改动有个问题，就是无法通过ASSERT测试出ev_allback1是否被调用了。ASSERT_EQ判断的东西是最后的结果。

Shaofa, 2018-07-17:
@elfplz ASSERT_EQ(g_var, 0x5B)就是用于判断ev_callback1是否被调用，被调用后g_var被赋值为0x5A，delayed_action2中对g_var++，这两个回调都被调用后g_var的值为0x5B

elfplz, 2018-07-17:
@Shaofa 那没有问题了，我理解得有问题。

closed 2018-07-17 by elfplz