alibaba/AliOS-Things #415: mqttapp添加BUILD_TYPE=debug参数后无法联网

LeoZhang-ATMEL, 2018-07-13:
**Describe the bug**

- mqttapp@developerkit 添加 **BUILD_TYPE=DEBUG**编译选项后，MQTT无法连接。
- 如果不添加BUILD_TYPE=DEBUG参数，可以联网成功，但如果等待mqtt session退出后，再次输入**mqtt**命令，程序仍旧会停止。

> $ aos make mqttapp@developerkit BUILD_TYPE=debug
> aos-cube version: 0.2.45
> Making config file for first time

**environment**
1. branch(version): July/11/2018, master: 42f8e47fb5baf713534d0f0b6cd833a2a87df789
2. Desktop OS(windows linux or macOS): Windows
3. example: mqttapp@developerkit BUILD_TYPE=debug

**Expected behavior**

**Screenshots**
aos make mqttapp@developerkit BUILD_TYPE=debug
[image]
aos make mqttapp@developerkit ，再等待第一次mqtt退出后，再次输入mqtt命令后，程序会停止在下图所示位置
[image]

**Additional context**

submarinezy, 2018-07-29:
另外，输入mqtt命令是什么操作？

LeoZhang-ATMEL, 2018-07-30:
不加DEBUG编译选项是可以跑的，直接aos make mqttapp@developerkit就可以，但要注意编译器一定要是5.4.1 20160919 (release)版本，我试过其他版本GCC也会卡在handshake，可以用arm-none-eabi-gcc --version查看。分支版本是master，commit id在帖子里有。mqtt命令就是在CLI输入mqtt就可以了。

submarinezy, 2018-07-31:
据说是这版更新是解决这个问题的：https://github.com/alibaba/AliOS-Things/pull/448。但我这边还是不行。

submarinezy, 2018-07-31:
出现类似这种打印，像是两条打印合并到一起输出的。!inf] io[<0I1>4 3c2_0c]o<nIn>e cSoenrnveecrt ecdo.nmnq t(t0 )c odninseccotn nseucctceeds.s。

closed 2022-03-01 by skylarCai