alibaba/AliOS-Things #459: Not able to bring up the board based on STM32F103VET6 after port Alios-Things.

mullife, 2018-08-02:
**Describe the bug**
failure to boot up board based on STM32F103VET6

**environment**
1. branch(version):   rel_1.3.2
2. Desktop OS(windows linux or macOS): Ubuntu 16.04
3. example: 

**Expected behavior**
Boot up successully for helloworld

**Screenshots**

**Additional context**
1. MCU: STM32F103VET6
2. Already refer to the following manual for porting and give the change accordingly fro Cortex-M3
    https://github.com/alibaba/AliOS-Things/wiki/Flash-and-KV-Storage-Porting-Guide
3.  Also fork Alios-Things and create a new branch for porting. please refer to the following link and branch.
    https://github.com/mullife/AliOS-Things/tree/rel_1.3.2_templedisplayer
    the new board name is "displayer".  all project is copied from b_l475e and make the changes accordingly for porting.
4. build successfully with "aos make mqttapp@displayer " in Ubuntu 16.04 and get the bin/hex.  But look like that failure to jump to entry_main and also no any log come out by printf

5. but check the same project with keil compiled with ARMCC and can boot up the board.

So for gcc compilation, any dependency on for STM32F103 porting and cause boot-up failure?
Already debug with a few days and but still failure. so hope that can get the support from here.

Thanks

mullife, 2018-08-04:
Nobody can share any information for checking in further?

librae8226, 2018-08-06:
Thanks for reporting! @mullife 
Would you please provide the boot up log?

mullife, 2018-08-07:
Unfortunatly there is no any log printed. if any log printed, and then i can check in further by myself.
look like main() functions is not launched so all of initlization is not done.

walker2048, 2018-08-08:
try this fork，https://gitee.com/walker2048/AliOS-Things/tree/F103RB/
UART and GPIO part is good to go。helloworld has CLI component，try `aos make helloworld@stm32f103rb-nucleo`。
my board is Nucleo-F103RB，using PA2 and PA3 for UART log。If your board is a Nucleo board，you can use it directly。

walker2048, 2018-08-08:
btw，don't set breakpoint at the main() in the `board/stm32f103rb-nucleo/Src/main.c`，the real entry point is main() in `platform\mcu\stm32l4xx_cube\aos\aos.c`.

mullife, 2018-08-08:
@walker2048 thanks for your information. will check it and come back.

mullife, 2018-08-08:
@walker2048 check out the branch of F103RB. seem like somes files in platform/mcu/stm32f1xx_cube/aos/ are missed, would you check it and commti as well? 
Thanks.

walker2048, 2018-08-08:
@mullife I've checked it.  Sorry about my mistack. 
It seem's like git ignored aos/* files both in the board/ and plarform/ .

chennanxu, 2018-08-09:
@walker2048 If I want to implement this code onto my own board, which part should I modify? Thanks

chennanxu, 2018-08-09:
@walker2048 thanks

walker2048, 2018-08-09:
@chennanxu 不好意思，看错了。。。

walker2048, 2018-08-09:
@mullife 
I had double check  the code，pull it from git again. 
Helloworld app with CLI complie success，and run on my nucleo board with no problem.

mullife, 2018-08-09:
@walker2048  Appreciate so much for your detail information. i will check it and feedback you.

mullife, 2018-08-09:
@walker2048 Greate job!!!!. check your porting and now i can get UART log from PA9 & PA10 on STM32F103C8T6 (my board is only working with USART1). I will check why is not working on STM32F103VET6 for my previous porting.

mullife, 2018-08-11:
@walker2048 As mentioned as above, based on your porting, chang port from USART 2 to USART 1(PA9 & PA10), it's able to print the log by printf from UART1.  but look like it's not able to get the input from UART1 and then CLI is not able to get char.
1. USART1_IRQHandler doesn't happen enven i input some char by chat.
2. Stuck in krhino_buf_queue_recv().

Have the following changes to switch the UART port from USART 2 to USART 1
1. hal_uart_stm32f1.h
   a). Change the definition
        #define PORT_UART1 0           /* USART1 */
        #define PORT_UART2 1           /* the port number of stand uart must be 0 */
   b). change the GPIO/CLK definition in this file for USART1.
2. Change the GPIO initialization in stm32f1xx_hal_msp.c for USART 1 GPIO.

Do you meet the same issue that CLI is not able to get the char for TX?
thanks.

mullife, 2018-08-12:
@walker2048 Yes, once receive the data, UART1 Intterruption will be trigged and HAL_UART_RxCpltCallback()  will be called then data will be sent to queue buffer by krhino_buf_queue_send()
Actually i have added the log in USART1_IRQHandler() and HAL_UART_RxCpltCallback(). but the log is never printed.

So check PA10(RX) state and it's always low even i put some char from laptop. so suppose that PA10 is not configured correctly. But check code with some time and didn't find any suspicion

it's sample to configured UART
1. hal_uart_init()
    UART1_TX_GPIO_CLK_ENABLE();
    UART1_RX_GPIO_CLK_ENABLE();
    UART1_CLK_ENABLE();
    HAL_NVIC_SetPriority(UART1_IRQn, 0, 1);
    HAL_NVIC_EnableIRQ(UART1_IRQn);
2. HAL_UART_MspInit()
    GPIO_InitStruct.Pin = GPIO_PIN_9|GPIO_PIN_10;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
    HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);

walker2048, 2018-08-12:
@mullife
The LOG function won't work in the IRQHandler functions.  I try to LOG in gpio_isr_handler(),  or uart irqhandler,  the log is never printed.  Why？ I didn't figure it out.
If you only modify the pinset of USART2 in `hal_uart_stm32f1.h`  , can it work?
Change `GPIO_PIN_2` to  GPIO_PIN_9 ,  `GPIO_PIN_3` to GPIO_PIN_10
I can try to use USART1 on pin PA9 and PA10 tomorrow.

walker2048, 2018-08-12:
@mullife
hal_uart_init()
UART1_TX_GPIO_CLK_ENABLE();       --- __HAL_RCC_GPIOB_CLK_ENABLE()    should be GPIOA
UART1_RX_GPIO_CLK_ENABLE();       --- __HAL_RCC_GPIOB_CLK_ENABLE()    should be GPIOA
UART1_CLK_ENABLE();
HAL_NVIC_SetPriority(UART1_IRQn, 0, 1);
HAL_NVIC_EnableIRQ(UART1_IRQn);

But in your case, it can print log, the GPIOA clock is enable.
I will check it tomorrow.

walker2048, 2018-08-13:
@mullife 
Modify the GPIO init config in `HAL_UART_MspInit()`, will fix this problem. 
 Create by `STM32CubeMX`:
```
    GPIO_InitStruct.Pin = GPIO_PIN_9;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
    HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);

    GPIO_InitStruct.Pin = GPIO_PIN_10;
    GPIO_InitStruct.Mode = GPIO_MODE_INPUT;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);
```

mullife, 2018-08-13:
@walker2048 Greate and change the mode to GPIO_MODE_INPUT and CLI is working fine now.

I thought i had this change for checking with GPIO_MODE_AF_INPUT which is the same as GPIO_MODE_INPUT

Anyway, it's greate support and Appreciate your help freely.

And now my STM32F103VET6 is also boot up.  The flash size of STM32F103C8T6 is too small for Alios-Things. So will have to STM32F103VET6

mullife, 2018-08-14:
@walker2048 Also share the porting changes by https://github.com/mullife/AliOS-Things/tree/rel_1.3.3_stm32f1xx_cube.
Thanks.

walker2048, 2018-08-14:
@mullife My pleasure.

closed 2022-03-01 by skylarCai