alibaba/AliOS-Things #467: esp8266配网的时间太短了？ 现在扫描才5 6秒

sofwu, 2018-08-07:
8266能配网的只会出现几秒，然后一直都不能配网了，这是什么问题？
[image]

librae8226, 2018-08-07:
请确认按键以后 do_awss_active 是否有生效？方便的话请贴出完整日志。

sofwu, 2018-08-07:
我要做灯具的闪烁配网没有按键的，上电后触发do_awss_active配网时间只有4~5s。

sofwu, 2018-08-07:
[Receive_20180807211705.txt](https://github.com/alibaba/AliOS-Things/files/2266497/Receive_20180807211705.txt)

sofwu, 2018-08-08:
我用的是rel1.3.2版本的，运行demo是linkkitapp。硬件是esp8266-12F，配网流程：我先给设备上电，然后操作app确认在闪烁再进行下一步，结果到了100%都没有配网成功。操作app时间大概10秒左右。

sofwu, 2018-08-08:
ps：我先操作app到确认闪烁那一步，然后给设备上电同时点下一步，这样是可以配网成功的。

librae8226, 2018-08-17:
请尝试如下步骤深入追查一下，看是否有帮助找到线索：
> 抓个空口包看一下手机有没有发出来，如果手机发出来了，设备上tcpdump一下，看看设备有没有收到包，如果设备也收到了，在sniffer_cb里面加个log，看看数据包有没有发给配网库

braveheart007, 2018-09-12:
@sifwu 请问用AliOS给ESP8266配网需要做适配吗？还是直接配网就可以？

librae8226, 2018-09-12:
@braveheart007 可以直接配网的

yunyinkeji, 2018-09-20:
@librae8226 esp8266上确实有这样的配网问题，我也碰到的，未解决。

librae8226, 2018-09-20:
尝试一下刚发布的2.0版，看是否还存在这个问题

Torchwoods, 2019-01-15:
ESP8266烧录alios 2.0.0 ,只有在复位后的几秒钟事件内按配网按键，才能够正常接收到WIFI的SSID和密码，如果是运行一段时间后，在按配网按键，模块无法接收到SSID和密码，这是什么原因？我用EMW3060测试是没有这种情况发生的。

xfdr0805, 2019-06-17:
> ESP8266烧录alios 2.0.0 ,只有在复位后的几秒钟事件内按配网按键，才能够正常接收到WIFI的SSID和密码，如果是运行一段时间后，在按配网按键，模块无法接收到SSID和密码，这是什么原因？我用EMW3060测试是没有这种情况发生的。

我也碰到了这个问题，你解决了吗

sofwu, 2019-06-17:
@xfdr0805 没有解决

closed 2022-03-01 by skylarCai