alibaba/AliOS-Things #485: 编译tinyengine_app@esp8266时mqtt模块报错

imairy, 2018-08-14:
编译tinyengine_app@esp8266时mqtt模块报错
同样的方式编译esp32正常
求助，谢谢
```bash
$ aos make tinyengine_app@esp8266
aos-cube version: 0.2.49
Check if required tools for esp8266 exist
Making config file for first time
processing components: tinyengine_app esp8266 platform/mcu/esp8266 vcall init auto_component
app_version:app-1.0.0-20180814.2312
framework/tinyengine/tinyengine.mk:59: BUILD_TYPE=release_log 
framework/tinyengine/tinyengine.mk:60: tiny_engine_BUILD_TYPE= 
framework/tinyengine/tinyengine.mk:61: HOST_ARCH=xtensa 
framework/tinyengine/tinyengine.mk:62: CONFIG_SYSINFO_PRODUCT_MODEL=ALI_AOS_ESP8266 
kernel_version:AOS-R-1.3.1
app_version:app-1.0.0-20180814.2312
framework/tinyengine/tinyengine.mk:59: BUILD_TYPE=release_log 
framework/tinyengine/tinyengine.mk:60: tiny_engine_BUILD_TYPE= 
framework/tinyengine/tinyengine.mk:61: HOST_ARCH=xtensa 
framework/tinyengine/tinyengine.mk:62: CONFIG_SYSINFO_PRODUCT_MODEL=ALI_AOS_ESP8266 
kernel_version:AOS-R-1.3.1
Build AOS Now
TOOLCHAIN_PATH=/Users/ZDF/Workspace/Git/AliOS-Things/build/compiler/gcc-xtensa-lx106/OSX/bin/
Compiling spiffs
Compiling alicrypto
Compiling auto_component
Compiling base64
Compiling cjson
Compiling cli
Compiling mqtt
In file included from ./include/aos/aos.h:18:0,
                 from framework/connectivity/mqtt/mqtt_client.c:41:
./include/aos/list.h:162:0: warning: "list_entry" redefined [enabled by default]
 #define list_entry(ptr, type, member) \
 ^
In file included from ./framework/protocol/linkkit/iotkit/base/utils/LITE-utils/src/lite-utils.h:33:0,
                 from framework/connectivity/mqtt/mqtt_client.c:26:
./framework/protocol/linkkit/iotkit/base/utils/LITE-utils/src/lite-list.h:248:0: note: this is the location of the previous definition
 #define list_entry(ptr, type, member) \
 ^
Compiling MQTTPacket
Compiling digest_algorithm
Compiling board_esp8266
Compiling fota
Compiling fota_download
Compiling download_http
Compiling socket_stand
Compiling fota_platform
Compiling fota_mqtt_transport
framework/fota/platform/mqtt/ota_transport.c: In function 'ota_mqtt_publish':
framework/fota/platform/mqtt/ota_transport.c:106:5: warning: passing argument 3 of 'mqtt_publish' discards 'const' qualifier from pointer target type [enabled by default]
     ret =  mqtt_publish(topic_name, 1, msg, strlen(msg) + 1);
     ^
In file included from framework/fota/platform/mqtt/ota_transport.c:17:0:
./framework/connectivity/mqtt/./mqtt_instance.h:125:5: note: expected 'void *' but argument is of type 'const char *'
 int mqtt_publish(char *topic, int qos, void *data, int len);
     ^
Compiling framework
framework/fota/platform/mqtt/version_report.c: In function 'active_publish':
framework/fota/platform/mqtt/version_report.c:102:5: warning: passing argument 3 of 'mqtt_publish' discards 'const' qualifier from pointer target type [enabled by default]
     ret =  mqtt_publish(topic_name, 1, msg, strlen(msg) + 1);
     ^
In file included from framework/fota/platform/mqtt/version_report.c:13:0:
./framework/connectivity/mqtt/./mqtt_instance.h:125:5: note: expected 'void *' but argument is of type 'const char *'
 int mqtt_publish(char *topic, int qos, void *data, int len);
     ^
Compiling tiny_engine
framework/tinyengine/bone-framwork/bone-engine/jse/be_jse_symbol.c: In function '_new_func_code_symbol':
framework/tinyengine/bone-framwork/bone-engine/jse/be_jse_symbol.c:469:17: warning: initialization discards 'const' qualifier from pointer target type [enabled by default]
     char* ptr = lex->src+charFrom;
[... 52 lines cut ...]
 ^
framework/tinyengine/bone-framwork/bone-engine/native-modules/mqtt/module_mqtt.c: In function 'module_handle_cb':
framework/tinyengine/bone-framwork/bone-engine/native-modules/mqtt/module_mqtt.c:181:13: error: 'for' loop initial declarations are only allowed in C99 mode
             for (int i = 0; i < sizeof(k_ipad); i++)
             ^
framework/tinyengine/bone-framwork/bone-engine/native-modules/mqtt/module_mqtt.c:181:13: note: use option -std=c99 or -std=gnu99 to compile your code
framework/tinyengine/bone-framwork/bone-engine/native-modules/mqtt/module_mqtt.c:227:22: error: redefinition of 'i'
             for (int i = 0; i < sizeof(out); ++i)
                      ^
framework/tinyengine/bone-framwork/bone-engine/native-modules/mqtt/module_mqtt.c:181:22: note: previous definition of 'i' was here
             for (int i = 0; i < sizeof(k_ipad); i++)
                      ^
framework/tinyengine/bone-framwork/bone-engine/native-modules/mqtt/module_mqtt.c:227:13: error: 'for' loop initial declarations are only allowed in C99 mode
             for (int i = 0; i < sizeof(out); ++i)
             ^
framework/tinyengine/bone-framwork/bone-engine/native-modules/mqtt/module_mqtt.c:233:17: error: 'for' loop initial declarations are only allowed in C99 mode
                 for (int j = 0; j < 2; ++j)
                 ^
make[1]: *** [out/tinyengine_app@esp8266/Modules/./framework/tinyengine/bone-framwork/bone-engine/native-modules/mqtt/module_mqtt.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [main_app] Error 2
[AliOS-Things] ERROR: "/" returned error code 2.
[AliOS-Things] ERROR: Command "/ U s e r s / Z D F / W o r k s p a c e / G i t / A l i O S - T h i n g s / b u i l d / c m d / o s x / m a k e   H O S T _ O S = O S X   T O O L S _ R O O T = / U s e r s / Z D F / W o r k s p a c e / G i t / A l i O S - T h i n g s / b u i l d   - e   - f   b u i l d / M a k e f i l e   t i n y e n g i n e _ a p p @ e s p 8 2 6 6" in "/Users/ZDF/Workspace/Git/AliOS-Things"
---
```

wuzhun1001, 2018-08-15:
当前官方发布版本TinyEngine仅支持MK3060，esp32devkitc，developerkit。其他版型没有验证过，如果开发者需要支持其他版型如esp8266，可自行解决相关问题。
贴图中的错误是因为编译器不支持c99，需要添加编译选项。

Jianlong0430, 2018-08-30:
@wuzhun1001 能否帮忙在README.md说明下对环境的依赖关系，如已验证过的开发板。关于README.md可以参考example下面的README.md。

Jianlong0430, 2018-08-30:
@imairy ,可查看framework/tinyengine/README.md，编译固件及下载里面有写到：当前已经验证通过的版型有 mk3060，esp32devkitc，developerkit。后续在example/tinyengine_app里面会有相应的README.md。

closed 2022-03-01 by skylarCai