alibaba/AliOS-Things #486: Developer Kit开发板环境配置MobaXterm检测不到任何端口（COM3）

grdaimap, 2018-08-15:
**Describe the bug**
硬件Developer Kit开发板；
按照https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-Windows-Environment-Setup的教程配置环境，USB-SERIAL CH340（COM3）显示未连接到计算机，多次拔插还是这样，MobaXterm检测不到端口，于是vscode不能烧录。

**environment**
1. branch(version): master
2. Desktop OS(windows linux or macOS):Windows10 1803
3. example: 见下

**Expected behavior**
[image]
这是教程里给的

**Screenshots**
这是我出错的截图
[image]
[image]
[image]

**Additional context**
这是vscode给的报错信息
aos-cube version: 0.2.50
[INFO]:target: helloworld@developerkit
[INFO]:Currently in aos_sdk_path: 'D:\exx\things\aliosthings'
aos-cube version: 0.2.50
scons: Reading SConscript files ...
[INFO]: Scons to upload!args:{'APPLICATION': 'helloworld', 'COMMAND': 'upload', 'BOARD': 'developerkit'}

[INFO]:target: helloworld@developerkit

[INFO]:Currently in aos_sdk_path: 'D:\exx\things\aliosthings'

[INFO]:upload_cmd: 'D:/exx/things/aliosthings/build/cmd/win32/st-flash.exe', host_os:'Win32'

[INFO]:image_path: 'D:\exx\things\aliosthings\out\helloworld@developerkit\binary\helloworld@developerkit.bin'

[INFO]:exec_cmd: '['D:/exx/things/aliosthings/build/cmd/win32/st-flash.exe', '--reset', 'write', 'D:\\exx\\things\\aliosthings\\out\\helloworld@developerkit\\binary\\helloworld@developerkit.bin', '0x08000000']'

st-flash 1.5.0
2018-08-15T12:07:11 WARN usb.c: Stlink usb device found, but unable to claim (probably already in use?)
[!] send_recv send request failed: LIBUSB_ERROR_NOT_FOUND
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_NOT_FOUND
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_NOT_FOUND
[!] send_recv STLINK_DEBUG_ENTER
Failed to enter SWD mode
[!] send_recv send request failed: LIBUSB_ERROR_NOT_FOUND
[!] send_recv STLINK_JTAG_WRITEDEBUG_32BIT
[ERROR]:firmware upload failed!

scons: done reading SConscript files.
scons: Building targets ...
scons: `.' is up to date.
scons: done building targets.

终端将被任务重用，按任意键关闭。

==END==

grdaimap, 2018-08-15:
问题暂时解决 :  )
我在QQ群里找到另外两份教程
把三个教程所有东西装上，烧录成功（但是端口是用COM4，COM3仍然灰色）
另外两个教程的文件在此 :  )

[AliOS Things Developer Kit Quick Start_V1.01.pdf](https://github.com/alibaba/AliOS-Things/files/2289343/AliOS.Things.Developer.Kit.Quick.Start_V1.01.pdf)
[WIN10x64系统安装AliOS Things编译环境.pdf](https://github.com/alibaba/AliOS-Things/files/2289344/WIN10x64.AliOS.Things.pdf)

walker2048, 2018-08-16:
@grdaimap 这个教程太老了，不合适用。
Developer Kit带stlink的，所以你只需要安装好stlink-v2驱动，上传固件（烧录），以及虚拟串口都有了。
一般情况下没必要另行接USB转串口模块。

librae8226, 2018-08-23:
Close according to above comments, please feel free to reopen if any further problem.

closed 2018-08-23 by librae8226

liuwei0514, 2018-09-17:
插 USB ST_link 那个Micro-USB 接口 不是USB2 for OTG接口 。
我是新手，这样解决了。