alibaba/AliOS-Things #487: ESP32 开机执行do_awss_active无法配网

fanday, 2018-08-15:
**Describe the bug**

**environment**
1. branch(version): 
2. Desktop OS: ubutnu 16.04
3. example: 

**Expected behavior**
代码如下：
`int application_start(int argc, char **argv)
{
#ifdef CSP_LINUXHOST
    signal(SIGPIPE, SIG_IGN);
#endif
#if AOS_ATCMD
    at.set_mode(ASYN);
    at.init(AT_RECV_PREFIX, AT_RECV_SUCCESS_POSTFIX,
            AT_RECV_FAIL_POSTFIX, AT_SEND_DELIMITER, 1000);
#endif

#ifdef WITH_SAL
    sal_init();
#endif
    aos_set_log_level(AOS_LL_DEBUG);

    netmgr_init();
    aos_register_event_filter(EV_WIFI, wifi_service_event, NULL);
#ifdef CONFIG_YWSS
    aos_register_event_filter(EV_KEY, linkkit_key_process, NULL);
    aos_register_event_filter(EV_YUNIO, cloud_service_event, NULL);
#endif

#if (defined (CONFIG_AOS_CLI) && defined (CONFIG_YWSS))
    aos_cli_register_command(&resetcmd);
    aos_cli_register_command(&ncmd);
#endif
    aos_task_new("netmgr", start_netmgr, NULL, 4096);

    do_awss_active();
    aos_loop_run();

    return 0;
}
`
期望开机能够配网成功

**Screenshots**
实际情况出现如下错误：
SSID: [Wxxxxx]
hint frame: offset:52, F, sn:292
src:bcec5debc2cf, bssid:fcd733ec7a14
16+0 [100] 294 F 3e2
16+1 [100] 295 F 123
16+2 [100] 296 F 1bf
16+3 [100] 297 F 220
16+4 [98] 299 F 280
16+5 [98] 29a F 317
16+6 [98] 29b F 3aa
16+7 [98] 29c F 426
16+8 [98] 29d F 4a7
24+0 [100] 2a1 F 3e3
24+0 [100] 2ac F 3e3
24+1 [100] 2ad F 119
24+2 [100] 2ae F 181
24+3 [100] 2af F 20f
24+4 [98] 2b1 F 29d
0+0 [100] 2b2 F 4e0
0+0 [100] 2b3 F 4e0
0+0 [100] 2b4 F 4e0
0+1 [98] 2b6 F 11c
0+2 [98] 2b7 F 18f
0+3 [98] 2b8 F 208
ssid auto-complete: Wxxxxx
0+4 [98] 2b9 F 28e
0+5 [96] 2bb F 337
                         uplimit [96] -> [98]
0+6 [98] 2bc F 3d8
                         uplimit [98] -> [99]
0+7 [99] 2bd F 445
                         uplimit [99] -> [100]
0+8 [100] 2be F 4cc
        enter try_to_sync_pos: data mismatch
        X+8 [1] 2be F 4cc
        exit try_to_sync_pos: group frame
[8]=cc, [8]=4cc
[24]=4a7, [8]=4cc
match=1, match_group=8, match_end=8
        X = 0, score=2, match=1, reason=3
8+0 [100] 2bf F 3e1
8+0 [100] 2c1 F 3e1
8+1 [100] 2c2 F 149
8+2 [100] 2c3 F 1c6
8+3 [100] 2c4 F 245
8+4 [98] 2c6 F 291
                         uplimit [98] -> [99]
8+5 [99] 2c7 F 33b
8+6 [99] 2c8 F 382
8+7 [99] 2c9 F 404
8+8 [99] 2ca F 4ac

62 65 65 62 65 65 65 65
65 65 65 65 63 63 63 63
64 64 64 62 62 62 62 62
64 64 64 62

␛[32m1c ␛[32m0f ␛[32m08 ␛[32m0e ␛[32m37 ␛[32m58 ␛[32m45 ␛[32m4c
␛[32m49 ␛[32m46 ␛[32m45 ␛[32m11 ␛[32m3b ␛[32m02 ␛[32m04 ␛[32m2c
␛[32m23 ␛[32m3f ␛[32m20 ␛[32m00 ␛[32m17 ␛[32m2a ␛[32m26 ␛[32m27
␛[32m19 ␛[32m01 ␛[32m0f ␛[32m1d

**Additional context**

librae8226, 2018-08-17:
请问是否有在 APP 端正确触发配网行为？

fanday, 2018-08-20:
目前是采用智能家居App扫描平台上的配网二维码，进入配网模式的，请问还需要额外的操作吗？

librae8226, 2018-08-24:
@fanday `do_awss_active();` 放在 start_netmgr 之前试试

librae8226, 2018-08-24:
另外如果四元组信息不对，可能导致密码无法解出，请确认四元组信息。

fanday, 2018-09-03:
@librae8226 Release 1.3.3版本可以正常配网，谢谢支持。

closed 2022-03-01 by skylarCai