alibaba/AliOS-Things #489: ESP8266编译错误

godchenmeng, 2018-08-16:
**Describe the bug**
Executing task: aos make helloworld@esp8266 <

aos-cube version: 0.2.50
Check if required tools for esp8266 exist
Making config file for first time
processing components: helloworld esp8266 platform/mcu/esp8266 vcall init auto_component
app_version:app-1.0.0-20180816.2301
kernel_version:AOS-R-1.3.1
app_version:app-1.0.0-20180816.2301
kernel_version:AOS-R-1.3.1
Build AOS Now
TOOLCHAIN_PATH=/Users/chenmeng/LoT/AliOS-Things/build/compiler/gcc-xtensa-lx106/OSX/bin/
Compiling alicrypto
Compiling auto_component
Compiling cli
Compiling board_esp8266
Compiling framework
Compiling hal
Compiling helloworld
Compiling kernel_init
Compiling vfs_device
Compiling newlib_stub
Compiling log
Compiling kv
Compiling xtensa
Compiling esp8266
Compiling rhino
Compiling vcall
Compiling espos
kernel/vcall/espos/platform/rhino/espos_queue.c: In function 'espos_queue_create':
kernel/vcall/espos/platform/rhino/espos_queue.c:53:12: warning: assignment makes integer from pointer without a cast [enabled by default]
     *queue = (espos_queue_t *)kqueue;
            ^
Compiling vfs
Compiling yloop
Making out/helloworld@esp8266/ld/eagle.app.v6.new.1024.app1.ld
Making out/helloworld@esp8266/libraries/alicrypto.a
Making out/helloworld@esp8266/libraries/auto_component.a
Making out/helloworld@esp8266/libraries/cli.a
Making out/helloworld@esp8266/libraries/board_esp8266.a
Making out/helloworld@esp8266/libraries/framework.a
Making out/helloworld@esp8266/libraries/hal.a
Making out/helloworld@esp8266/libraries/helloworld.a
Making out/helloworld@esp8266/libraries/kernel_init.a
Making out/helloworld@esp8266/libraries/vfs_device.a
Making out/helloworld@esp8266/libraries/newlib_stub.a
Making out/helloworld@esp8266/libraries/log.a
Making out/helloworld@esp8266/libraries/kv.a
Making out/helloworld@esp8266/libraries/xtensa.a
Making out/helloworld@esp8266/libraries/esp8266.a
Making out/helloworld@esp8266/libraries/rhino.a
Making out/helloworld@esp8266/libraries/vcall.a
Making out/helloworld@esp8266/libraries/espos.a
Making out/helloworld@esp8266/libraries/vfs.a
Making out/helloworld@esp8266/libraries/yloop.a
Making helloworld@esp8266.elf

Making helloworld@esp8266.bin
Making helloworld@esp8266.hex
no definite address hint,using default mem configuration

                        AOS MEMORY MAP
|=================================================================|
| MODULE                                   | ROM       | RAM      |
|=================================================================|
| newlib_stub                              | 0         | 277      |
| kv                                       | 0         | 3622     |
| vcall                                    | 0         | 2686     |
| kernel_init                              | 0         | 782      |
| rhino                                    | 0         | 17957    |
| framework                                | 0         | 382      |
| vfs                                      | 0         | 3367     |
| libgcc                                   | 0         | 1196     |
| libmain                                  | 0         | 30187    |
| libpp                                    | 0         | 62406    |
| esp8266                                  | 0         | 9072     |
| board_esp8266                            | 0         | 305      |
| libcirom                                 | 0         | 29505    |
| yloop                                    | 0         | 1396     |
| libnet80211                              | 0         | 45787    |
| xtensa                                   | 0         | 6863     |
| libhal                                   | 0         | 37       |
| log                                      | 0         | 533      |
| espos                                    | 0         | 1138     |
| helloworld                               | 0         | 236      |
| cli                                      | 0         | 6612     |
| hal                                      | 0         | 207      |
| libwpa                                   | 0         | 17862    |
| liblwip                                  | 0         | 39438    |
| libcrypto                                | 0         | 14512    |
| libphy                                   | 0         | 48619    |
| *fill*                                   | 0         | 1604     |
|=================================================================|
| TOTAL (bytes)                            | 0         | 346588   |
|=================================================================|
/Users/xxxx/LoT/AliOS-Things/build/cmd/osx/dash: 1: python2: not found
mv: eagle.app.flash.bin: No such file or directory
Build complete
Making .gdbinit
最后无法找到文件eagle.app.flash.bin，最后binary文件夹下没有文件xxxxx@esp8266-0x1000.bin
**environment**
1. branch(version):  最新版本
2. Desktop OS(windows linux or macOS): macOS
3. example: 

**Expected behavior**

**Screenshots**

**Additional context**

librae8226, 2018-08-17:
这里面有两个问题：
1. Mac 下的 python 没有 python2 这个东西，其实它是个软链接。
    解决方法：
    1) 可以直接把它链接到 python2.7
    2) 修改 `platform/mcu/esp8266/gen_crc_bin.mk`: `python2` -> `python`
2. 不会生成 xxx@esp8266-0x1000.bin
    解决方法：修改 `site_scons/scons_upload.py`，如下
````
diff --git a/site_scons/scons_upload.py b/site_scons/scons_upload.py
index ec0511b80d..417db4c97e 100755
--- a/site_scons/scons_upload.py
+++ b/site_scons/scons_upload.py
@@ -101,7 +101,7 @@ def upload_esp8266(target, aos_path):
         error('Unsupported Operating System!')
     cmd = 'esptool.py'
     port = miniterm.ask_for_port()
-    image_path = os.path.join(os.getcwd(), 'out', target, 'binary', target + '-0x1000.bin').replace('\\', '/')
+    image_path = os.path.join(os.getcwd(), 'out', target, 'binary', target').replace('\\', '/')^M
     boot_bin_path = os.path.join(aos_path, 'platform/mcu/esp8266/bsp/boot_v1.7_921600.bin').replace('\\', '/')
     init_bin_path = os.path.join(aos_path, 'platform/mcu/esp8266/bsp/esp_init_data_default.bin').replace('\\', '/')
     blank_bin_path = os.path.join(aos_path, 'platform/mcu/esp8266/bsp/blank.bin').replace('\\', '/')
````

chenyulun, 2019-01-30:
@librae8226 
后面的` ^M`什么作用？

xiaowenxia, 2019-01-30:
@chenyulun 这是换行符的问题，没有什么影响的。