alibaba/AliOS-Things #491: 建议增加多国语言支持及环境配置功能

walker2048, 2018-08-20:
**越来越多非专业行业的使用者和学生接触到AliOS-Things，为方便广大初学者，提出以下建议**
1. 增加多国语言支持
很多初学者没有相关环境搭建、维护知识，同时也不具备C语言开发常识和系统常识。在环境搭建方面，以及项目目录结构，配置方面，也有很多疑问。增加多国语言支持，一是可以增强系统的易用性，二是方便日后国外用户本地化。
2. 增加环境检测功能（或者叫系统配置脚本）
AliOS-Things包含的内容是相当广的，从单片机功能实现，到网络、加密、文件系统、传感器驱动等。对于初学者来说，这个系统的复杂性太大了，不容易理清楚对应的目录，组件，以及内容。
2.1 环境配置功能说明
第一步，确认开发者使用的开发板。业余开发者手头上的开发板也许就那么一两块，他们手头上的板子，当前AliOS-Things支持情况如何，并不是很清楚。确认开发者使用的开发板后，提出相关环境搭建及系统功能补充说明（例如是否能在当前系统顺利编译，上传等等）。
第二步，根据开发者的开发板，确认当前安装环境缺什么，例如python环境是否准确有效，是否有多版本python环境存在的可能性，以及环境配置的一些建议。不能自动化安装环境无所谓，能提供一些建议就行。
第三步，根据开发者的开发板，对SDK进行优化配置（在保留master主线更新功能情况下，删除多余开发板的相关代码）

librae8226, 2018-08-20:
谢谢建议!
多语言希望哪些地方先支持呢？目前网站和wiki文档大部分有中英版本，开发工具还没有支持i18n。

walker2048, 2018-08-20:
@librae8226 可以考虑先从最基本的编译，上传等提示信息开始吧。
网站和wifi的教程版本相差有点大。也有点混乱。
也许是玩web开发玩多了，觉得i18n是必备功能吧。。

closed 2022-03-01 by skylarCai