alibaba/AliOS-Things #507: ESP8266的APP烧录之后无法运行

lindabang, 2018-08-25:
**Describe the bug**
ESP8266主板的APP烧录之后，每次加电串口显示“OS SDK ver: 2.0-dev(d03acf5) compiled @ Feb 23 2018 19:45:05”，然后就没有输出了。
我测试了helloworld和linkkitapp，以及其他的APP都是同样的结果。

**environment**
1. branch(version): master (commit:ba6380fdf32ef471b3e88af0df4c7934797b4629）
2. Desktop OS(windows linux or macOS):Ubuntu 18.04
3. example: aos make helloworld@esp8266;然后烧录 out/helloworld@esp8266/binary/helloworld@esp8266.bin及其他文件

**Expected behavior**
调用helloworld.c中的application_start函数，并打印消息

**Screenshots**
2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

OS SDK ver: 2.0-dev(d03acf5) compiled @ Feb 23 2018 19:45:05

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

OS SDK ver: 2.0-dev(d03acf5) compiled @ Feb 23 2018 19:45:05

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

OS SDK ver: 2.0-dev(d03acf5) compiled @ Feb 23 2018 19:45:05

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

OS SDK ver: 2.0-dev(d03acf5) compiled @ Feb 23 2018 19:45:05

**Additional context**

lindabang, 2018-08-25:
test@test-VirtualBox:~$ mv aos aos-esp

test@test-VirtualBox:~$ mv aos-aliyun aos

test@test-VirtualBox:~/aos$ aos make helloworld@esp8266
aos-cube version: 0.2.51
Check if required tools for esp8266 exist
Build AOS Now
TOOLCHAIN_PATH=/home/test/aos/build/compiler/gcc-xtensa-lx106/Linux64/bin/
Compiling auto_component
Making out/helloworld@esp8266/libraries/auto_component.a
Making helloworld@esp8266.elf

Making helloworld@esp8266.hex
Making helloworld@esp8266.bin
no definite address hint,using default mem configuration

                        AOS MEMORY MAP                            
|=================================================================|
| MODULE                                   | ROM       | RAM      |
|=================================================================|
| newlib_stub                              | 0         | 277      |
| kv                                       | 0         | 3622     |
| vcall                                    | 0         | 2686     |
| kernel_init                              | 0         | 782      |
| rhino                                    | 0         | 17957    |
| framework                                | 0         | 382      |
| vfs                                      | 0         | 3371     |
| libgcc                                   | 0         | 1196     |
| libmain                                  | 0         | 30187    |
| libpp                                    | 0         | 62406    |
| esp8266                                  | 0         | 9103     |
| board_esp8266                            | 0         | 305      |
| libcirom                                 | 0         | 29505    |
| yloop                                    | 0         | 1396     |
| libnet80211                              | 0         | 45787    |
| xtensa                                   | 0         | 6863     |
| libhal                                   | 0         | 37       |
| log                                      | 0         | 533      |
| espos                                    | 0         | 1138     |
| helloworld                               | 0         | 345      |
| cli                                      | 0         | 6612     |
| hal                                      | 0         | 207      |
| libwpa                                   | 0         | 17862    |
| liblwip                                  | 0         | 39438    |
| libcrypto                                | 0         | 14512    |
| libphy                                   | 0         | 48619    |
| *fill*                                   | 0         | 1596     |
|=================================================================|
| TOTAL (bytes)                            | 0         | 346724   |
|=================================================================|
bin crc: 117a7c93
Build complete
Making .gdbinit
test@test-VirtualBox:~/aos$

librae8226, 2018-08-30:
如何烧录的呢？看起来像烧录地址不对，或是flash分区配置有问题

Jianlong0430, 2018-08-30:
@lindabang 需要提供下烧录方法和步骤，尤其是烧录地址这块。该板子也可以使用aos upload helloworld@esp8266。 具体代码实现可参考：site_scons/scons_upload.py里面的8266 upload实现。

lindabang, 2018-09-03:
谢谢各位，烧录方式如下，但我感觉不是烧录地址不对，因为串口已经输出“OS SDK ver: 2.0-dev(d03acf5) compiled @ Feb 23 2018 19:45:05”，应该是已经运行了我烧录的程序了
[image]

librae8226, 2018-09-03:
> “OS SDK ver: 2.0-dev(d03acf5) compiled @ Feb 23 2018 19:45:05”
这个日志是 esp8266 的静态库 libmain.a 里面打印的，看起来还没有正确跳到OS。
你截图的烧录地址和正常的不太一样：
请参考：site_scons/scons_upload.py: upload_esp32()
```
        '0x0', boot_bin_path,
        '0x3fc000', init_bin_path,
        '0x3fe000', blank_bin_path,
        '0x1000', image_path
```

xuhongv, 2018-09-05:
@lindabang 烧录工具的 ```SPI MODE``` 选择为 ``` DIO``` .

angle600, 2019-09-30:
>SPI Flash Size & Map: 32Mbit(512KB+512KB)

------------------------------------
好像只支持32Mbit 1024 + 1024

closed 2022-03-01 by skylarCai