alibaba/AliOS-Things #510: linkkitapp@esp8266

wangjiangck, 2018-08-26:
**Describe the bug**
我基于例子linkkitapp，修改物的三要素与tsl，添加了一点测试代码：设置了几个属性值并上报到云端。手机APP使用公版，用APP扫描产品发布-->设备配网二维码进行配网。发现有时候配网成功，但有时候配网不成功，配网成功时点击“开始使用”按钮后，界面显示“加载中”并一直卡在这个界面。观察log也发现一些错误

**environment**
1. branch(version): v1.3.3
2. Desktop OS(windows linux or macOS):linux 
3. example: linkkitapp

**Expected behavior**
期望可以正确配网并连接到阿里云，并能在公版APP上看到上传的属性值

**Screenshots**
[image]
[image]
[image]
[test.txt](https://github.com/alibaba/AliOS-Things/files/2321419/test.txt)

**Additional context**

realonlyme, 2018-08-30:
这个是2个问题么?一个是配网时APP的问题，注意的是APP扫的码一定要和烧录至设备端的ProductID一致；

另外一个上传属性值是你设备端上行？还是云端下行？

wangjiangck, 2018-09-01:
是设备端上行啊，APP扫描的码和烧录至设备端的ProductID是一致的

braveheart007, 2018-09-12:
请问aos make linkkitapp@esp8266是不是得做适配之后才能用呢？ @wangjiangck

mq9494, 2019-01-25:
问题解决了吗。我这里配网成功，但是绑定失败。不知道怎么办，求助啊

closed 2022-03-01 by skylarCai