alibaba/AliOS-Things #536: developerkit JS疑问

saybabi, 2018-09-02:
**Describe the bug**
使用be-cli 编译生成bin
~/AliOS-Things/framework/tinyengine/samples/app/flowled~

异常: 硬件只有led3亮， led1, led2均不亮
**environment**
1. branch(version): 
2. Desktop OS(windows linux or macOS):
3. example: 

**Expected behavior**

**Screenshots**

**Additional context**

wuzhun1001, 2018-09-04:
你好，这是因为sample里面的例子使用的develoerkit老的板子（该板子上有5个LED灯）。
对于您手头上的develoerkit只有三个LED灯，如果需要控制LED灯，需要关注LED灯对应的具体GPIO引脚。即board_config.json中的配置。
方法：1） 通过原理图找到LED灯的具体引脚，即PB6，PE3，PD15。
           2）然后通过board/developerkit/hal/hal_gpio_stm32l4.h找到管脚的对应GPIO。（以developerkit为例，PB6=GPIO22 ， PE3=GPIO67 ，PD15=GPIO63）。
           3）在board_config.json中增加配置，如：
{
	"GPIO":[
		{
		"id":"led1",
		"port":22,
		"dir":0,
		"pull":1
		}，
              {
		"id":"led2",
		"port":67,
		"dir":0,
		"pull":1
		}，
              {
		"id":"led3",
		"port":63,
		"dir":0,
		"pull":1
		}
	]
}
  4: 最后在js文件中通过GPIO.write()去控制即可。

saybabi, 2018-09-04:
@wuzhun1001  
其实board_config.json中的id属性是不是自定义命名的？ 只要使用GPIO.open('idXXX')系统会找到对应的id值是吧？

wuzhun1001, 2018-09-04:
是的，id只是一个命名，主要是port，port对了就会控制底层的硬件。

saybabi, 2018-09-04:
明白，谢谢 :)

closed 2018-09-04 by saybabi