alibaba/AliOS-Things #546: [投票] 选择你最喜爱的开发环境，持续有效

librae8226, 2018-09-04:
看到这么多 issue，我们是很开心的，越来越多人开始关心我们的项目，对于我们来讲这也是一份更重的责任。其中每一个 issue 我们都有认真看，并且记录下来跟踪，谢谢大家的耐心等待。让我们从最基础的问题开始，一起改进提高吧！

这个投票的目的是希望能帮助大家有效解决目前物联网设备侧开发最常见的问题之一：**开发环境搭建**
投票方式很简单：为便于统计，在下方评论里，为你习惯的、喜欢的、或是希望使用的开发环境点赞👍

如果下方评论中没有列出你喜欢的开发方式，也可以回复一条新的。
当然，如果以下方式都熟悉，也可以都点赞。

- Keil MDK
- IAR Embedded Workbench
- 命令行: Make 等 Command Line 方式的构建工具
- Visual Studio Code: AliOS Studio
- 虚拟镜像: Docker
- 虚拟镜像: VirtualBox
- 其他

librae8226, 2018-09-04:
### Keil MDK
- 主机: Windows
- 软件: [Keil](http://www.keil.com)
- 支持情况: 开发中的功能。可通过脚本从 Makefile 生成 Keil 工程文件

librae8226, 2018-09-04:
### IAR Embedded Workbench
- 主机: Windows
- 软件: [IAR](https://www.iar.com)
- 支持情况: ，开发中的功能。可通过脚本从 Makefile 生成 IAR 工程文件

librae8226, 2018-09-04:
### 命令行: Make 等 Command Line 方式的构建工具
- 主机: Linux, Mac, Windows
- 软件: 各种 Terminal，GCC 系列工具链，GDB 等调试工具
- 支持情况: 默认支持。依赖 python 2.7 和 [aos-cube](https://pypi.org/project/aos-cube)
- 备注: 该项也包括所有以命令行工具为基础的开发环境，比如自建的 Sublime Text + zsh + make 环境等

librae8226, 2018-09-04:
### Visual Studio Code: AliOS Studio
- 主机: Linux, Mac, Windows
- 软件: [VS Code](https://code.visualstudio.com/), [alios-studio](https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-Studio)
- 支持情况: 默认支持。依赖 python 2.7 和 [aos-cube](https://pypi.org/project/aos-cube)

librae8226, 2018-09-04:
### 虚拟镜像: Docker
- 主机: Linux, Mac, Windows
- 软件: [Docker](https://www.docker.com/)
- 支持情况: 暂无支持。可参考贡献者 Jack 提供的版本 [lanjackg2003/alios-things-docker](https://github.com/lanjackg2003/alios-things-docker)

librae8226, 2018-09-04:
### 虚拟镜像: VirtualBox
- 主机: Linux, Mac, Windows
- 软件: [VirtualBox](https://www.virtualbox.org)
- 支持情况: 暂无支持

angengzhen, 2018-09-04:
KEIL

librae8226, 2018-09-04:
### Web IDE
- 主机: Linux, Mac, Windows
- 软件: 主流浏览器，Web IDE 依赖的本地工具(用于烧录/调试)
- 支持情况: 开发中
- 备注: web 端默认包含了需要的开发环境

iDea2016su, 2018-09-11:
IDEA

iDea2016su, 2018-09-11:
Visual Studio + Visual GDB
这个搭配已经支持了很多厂商的器件
利用了Visual Studio的高级功能
缺点是头文件没有精简，编译产生的目标文件太大

Changes729, 2019-03-19:
VSCode + PlatformIO + OpenOCD，
首先，通过VSCode实现Linux、Windows、Mac夸平台，轻量化，Git工具集成，好处不用多说了。
使用Platform实现编译和烧写，库比较全，用起来也挺方便（缺点这里略过）
最后使用OpenOCD实现Debug（PlatformIO就是用OpenOCD的，只是人家封装了一层变成收费了）

legenddcr, 2019-03-27:
> ### Web IDE: 敬请期待
> * 主机: Linux, Mac, Windows
> * 软件: 主流浏览器，Web IDE 依赖的本地工具(用于烧录/调试)
> * 支持情况: 开发中
> * 备注: web 端默认包含了需要的开发环境

支持AliOS-Things开发的Web IDE已经上线：https://hacklab.aliyun.com

karlzheng, 2019-09-19:
Mac / Linux : 命令行: Vim + Make + GCC

xgarett, 2019-10-31:
Keil MDK 
    主要是windows开发环境.keil用的人应该是最多的.也是适合新人学习.

Visual Studio Code: AliOS Studio  
    纯软的开发思维吧.跨平台比较好整.但是估计很多做嵌入式起步的人都不是很适应.........

ithinco, 2019-12-18:
Docker 挂载项目目录，VS Code 编辑后 Hot Reload

closed 2022-03-01 by skylarCai