alibaba/AliOS-Things #548: 成功编译了《Hello World》工程并且烧录到8266板子，但是发现串口打印信息不对劲？？

xuhongv, 2018-09-05:
**bug**描述：正常来说，不是应该显示打印 ```hello  world```吗？

**环境：**
1. branch(version): commit e8cb9ec7db73dc37036e06b758bc003d74106eef (HEAD -> master, origin/master, origin/HEAD)

2. Desktop OS(windows linux or macOS):  Ubuntu
3. example: 《Hello World》

**串口打印**
```
2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000

[11:16:52.851] OS SDK ver: 2.0-dev(d03acf5) compiled @ Feb 23 2018 19:45:05
[11:16:52.898] Wifi init success!!
mode : softAP(6a:c6:3a:b2:73:c8)
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
bcn 100
trace config close!!!
[11:16:55.297] [WiFi] event 8
[11:17:02.578] [WiFi] event 8
[11:17:03.748] [WiFi] event 8
[11:17:03.856] [WiFi] event 8
[11:17:05.238] [WiFi] event 8
[11:17:05.295] [WiFi] event 8
[11:17:05.415] [WiFi] event 8

```

**截图**

[image]

-----------------------------------------------------
**烧录固件地址**

|固件|地址|
|:-------:|:------------- |
| AliOS-Things\platform\mcu\esp8266\bsp\esp_init_data_default.bin | 0x3FC000 |
| AliOS-Things\platform\mcu\esp8266\bsp\blank.bin | 0x3FE000 |
| AliOS-Things\platform\mcu\esp8266\bsp\boot_v1.7_921600.bin | 0x0 |
| AliOS-Things\out\helloworld@esp8266\binary\helloworld@esp8266.bin | 0x1000 |

librae8226, 2018-09-07:
谢谢 @xuhongv 细致的问题说明，很抱歉给您带来的混淆。
请参考[这个提交](https://github.com/librae8226/AliOS-Things/commit/4a065444e6b65f5a1eff6000c2aa4a53d5a38cde)，按此修改烧录以后会得到期望的结果。

librae8226, 2018-09-07:
等下一版发布吧

xuhongv, 2018-09-07:
好吧。thanks

closed 2022-03-01 by skylarCai