alibaba/AliOS-Things #551: Be-cli项目 MQTT 订阅疑问

saybabi, 2018-09-05:
**Describe the bug**
问题:
1. 发现 MQTT.subscribe 中回调函数中的 `topic` 跟 `msg` 似乎对应值相反了。
2. 我想问下`1970-01-01 00:22:47.320 WARN  [JSE     752] s->flags = 0x5` 这个内容是正常的吗？
~~~
device.subscribe = function(){
    console.log('setting subscribe...');
    MQTT.subscribe("/xxx", function(topic, msg){
        console.log('topic');
        console.log(topic);
        console.log('msg');
        console.log(msg);
    });
};
~~~

日志反馈
~~~
[dbg] iotx_mc_cycle(1288): PUBLISH
[dbg] iotx_mc_handle_recv_PUBLISH(1122): delivering msg ...
[dbg] iotx_mc_deliver_message(877): topic be matched
BoneEngine > topic
1970-01-01 00:22:47.320 WARN  [JSE     752] s->flags = 0x5

BoneEngine >
BoneEngine > msg
BoneEngine > /sys/a1UbpRptz/demo/thing/service/property/set
1970-01-01 00:22:52.550 DEBUG [MODULE_TIMERS 41] JS handle =536971520
~~~

**environment**
1. branch(version): 
2. Desktop OS(windows linux or macOS):
3. example: 

**Expected behavior**

**Screenshots**

**Additional context**

saybabi, 2018-09-05:
@wuzhun1001  大佬，有空的话指点下我。

wuzhun1001, 2018-09-06:
你好，
1  这里mqtt的MQTT.subscribe("/xxx", function(topic, msg)中的 function(topic, msg)参数是错误的，应该是 function(msg，topic)。
我们会更新API文档，让这些API更加清晰，也可以参考里面的sample js。
2 这里的打印是s->flags = 0x5是一个警告，这是因为msg一个json对象，我们这里的js语法暂时不支持直接console.log一个对象，需要具体到成员。所以会有这个警告，不影响功能。

wuzhun1001, 2018-09-06:
参考samples里面的mqtt.js例子，可以看到json对象可以使用 JSON.stringify(msg)打印。

saybabi, 2018-09-06:
好的 👍

closed 2018-09-06 by saybabi