alibaba/AliOS-Things #569: 我已经成功esp8266的串口一作为日志打印，《linkkitapp》竟然还会有部分数据日志打印？？

xuhongv, 2018-09-12:
**bug**描述：不断打印 ```chan1 `chan2```的bug，而其他调用`printf()`则不会，我怎么屏蔽这个不断打印？我把 `uart1` 作为日志打印，竟然还会打印，过了三分钟之后，就不会打印了，奇怪。。。这边项目需要的是串口通讯。。。。紧急。。。

**环境：**
1. branch(version): commit e8cb9ec7db73dc37036e06b758bc003d74106eef (HEAD -> master, origin/master, origin/HEAD)

2. Desktop OS(windows linux or macOS):  Ubuntu
3. example: 《linkkitapp》

**串口打印**
```
chan 6
chan 9
chan 11
chan 12
chan 1
chan 6
chan 11
chan 1
chan 2
chan 3
chan 4
chan 5
chan 6
chan 7
chan 8
chan 9
chan 10
chan 11
chan 12
chan 13
chan 1
chan 2
chan 6
chan 9
chan 11
chan 12
chan 1
chan 6
chan 11
chan 1
chan 2
chan 3
chan 4
chan 5
chan 6
chan 7
chan 8
chan 9

```

librae8226, 2018-09-12:
是 framework/ywss4linkkit/lib/xtensa/esp8266/ywss4linkkit.a 这里面调用 HAL_Printf 打印出来的
可以跟一下你的代码里具体 HAL_Printf 是调用到哪个串口去了

xuhongv, 2018-09-12:
- 最后发现了问题：不知道在哪里调用了` uart.c`的 `hal_uart_send()`，我在源文件注释代码如下即可 去掉频繁打印了：

```
int32_t hal_uart_send(aos_uart_dev_t *uart, const void *data, uint32_t size, uint32_t timeout)
{
    int i = 0;
    char *pdata = (char *)data;

    for (i = 0; i < size; i++)
    {
        uart0_write_char(pdata[i]);
    }

    return 0;
}

```

xuhongv, 2018-09-12:
@librae8226  我把 8266 里面的 `hal_uart_send()`  去掉，发现报错了如下！存在2个方面原因，要么是：`esp8266/libraries/newlib_stub.a` 这个文件调用去掉，要么是`utility/libc/newlib_stub.c`去掉调用！

最后发现：
```
out/linkkitapp@esp8266/libraries/newlib_stub.a(newlib_stub.o):(.literal._write_r+0x8): undefined reference to `hal_uart_send'
out/linkkitapp@esp8266/libraries/newlib_stub.a(newlib_stub.o): In function `_write_r':
/media/sf_Alios_Esp8266/AliOS-Things/utility/libc/newlib_stub.c:80: undefined reference to `hal_uart_send'
/media/sf_Alios_Esp8266/AliOS-Things/utility/libc/newlib_stub.c:80: undefined reference to `hal_uart_send'
```

xuhongv, 2018-09-12:
- 对于`esp8266` 有2个串口，分别是串口0和串口一，而串口一仅仅可以做收数据。所以我们一般用串口0作为与其他设备串口通讯的串口，而串口一作为打印系统日志；

- 现在我已经成功地把系统日志打印到了串口一，只是把` printf() `打印到串口一，而通过串口0发送数据是调用 `uart0_write_char()` ;
 

- 上面提到，我们只能通过`uart0_write_char()` 发送数据到其他设备，而 `esp8266` 适配层的 `hal_uart_send()` 是调用 `uart0_write_char()` 发送的！

- 而此个 bug 反复出现 `chan1 chan5`，正是反复调用此函数。有什么办法可以不要发送，或者调用  printf() `打印。

walker2048, 2018-09-13:
Linkitapp，调用了一键配网库，这里输出频道列表的，应该是一键配网那边的。

xuhongv, 2018-09-13:
两边都不开源的。只能治标不治本如下修改：

```
int32_t hal_uart_send(aos_uart_dev_t *uart, const void *data, uint32_t size, uint32_t timeout)
{

    int i = 0;
    char *pdata = (char *)data;
    for (i = 0; i < size; i++)
    {
        if (uart->port == 1)
            uart0_write_char(pdata[i]);
        else
            uart1_write_char(pdata[i]);
    }

    return 0;
}
```

closed 2022-03-01 by skylarCai