alibaba/AliOS-Things #574: toolchain auto-install error: can not remove temp folder tmp_be, please remove it manually

SuperCoderMan, 2018-09-13:
**Describe the bug**
**environment**
1. branch(version): master
2. Desktop OS(windows linux or macOS):win10+VS Code
3. 现象: 
创建新的工程后，首次编译时出现：
···
aos-cube version: 0.2.57
Check if required tools for esp8266 exist
toolchain gcc-xtensa-lx106 missing, start download ...
https://gitee.com/alios-things/gcc-xtensa-lx106-win32.git -> build/compiler/gcc-xtensa-lx106/Win32
Cloning into 'gcc-xtensa-lx106'...
remote: Enumerating objects: 966, done.
remote: Counting objects: 100% (966/966), done.
remote: Compressing objects: 100% (560/560), done.
remote: Total 966 (delta 388), reused 966 (delta 388)
Receiving objects: 100% (966/966), 86.52 MiB | 2.12 MiB/s, done.
Resolving deltas: 100% (388/388), done.
Checking connectivity... done.
Checking out files: 100% (916/916), done.
download toolchain gcc-xtensa-lx106 succeed
toolchain auto-install error: can not remove temp folder tmp_be, please remove it manually
Making config file for first time
···
文件夹tmp_be为该过程中自行创建，然后一直停在Making config file for first time，手动删除再次编译，一样停在Making config file for first time。

librae8226, 2018-09-13:
抱歉，windows 下 esp8266 目前无法编译。Windows下请使用虚拟机或Docker镜像，或是直接使用linux/mac机器开发

SuperCoderMan, 2018-09-17:
好的，多謝！

closed 2018-09-17 by SuperCoderMan