alibaba/AliOS-Things #598: aos make linkkitapp@mk3239编译有问题，请求协助下

QiJiaHai, 2018-09-19:
**Describe the bug**

**environment**
1. branch(version): 
2. Desktop OS(windows linux or macOS):
3. example: 

**Expected behavior**

**Screenshots**

**Additional context**

QiJiaHai, 2018-09-19:
主分支不行，然后切到1.3.3还是不行 windows电脑开发的

QiJiaHai, 2018-09-19:
> Executing task: aos make linkkitapp@mk3239 <

aos-cube version: 0.2.59
Check if required tools for mk3239 exist
Build AOS Now
TOOLCHAIN_PATH=C:/Users/qijiahai/Desktop/wulian/alios-things-AliOS-Things-rel_1.3.3/AliOS-Things/build/compiler/gcc-arm-none-eabi/Win32/bin/
Components: linkkitapp board_mk3239 stm32f4xx vcall kernel_init auto_component linkkit_sdk iotx-hal netmgr framework libywss uOTA cjson cli armv7m newlib_stub rhino mbedtls digest_algorithm Lib_SPI_Flash_Library net STM32F4xx_Peripheral_Drivers activation chip_code kv yloop hal alicrypto STM32F4xx_Peripheral_Libraries log vfs vfs_device
Skipping building bootloader due to \"total\" is not set

process_begin: CreateProcess(NULL, sh ././framework/protocol/linkkit/sdk/iotx_sdk_make_win.sh out/linkkitapp@mk3239/libraries ./example/linkkitapp, ...) failed.
make (e=2): 系统找不到指定的文件。
make.exe[1]: *** [build/aos_target_build.mk:262: out/linkkitapp@mk3239/libraries/libiot_sdk.a] Error 2
make.exe[1]: *** Waiting for unfinished jobs....
make.exe: *** [build/Makefile:265: main_app] Error 2
[AliOS-Things] ERROR: "C" returned error code 2.
[AliOS-Things] ERROR: Command "C : / U s e r s / q i j i a h a i / D e s k t o p / w u l i a n / a l i o s - t h i n g s - A l i O S - T h i n g s - r e l _ 1 . 3 . 3 / A l i O S - T h i n g s / b u i l d \ c m d / w i n 3 2 / m a k e . e x e   H O S T _ O S = W i n 3 2   T O O L S _ R O O T = C : / U s e r s / q i j i a h a i / D e s k t o p / w u l i a n / a l i o s - t h i n g s - A l i O S - T h i n g s - r e l _ 1 . 3 . 3 / A l i O S - T h i n g s / b u i l d   - e   - f   b u i l d / M a k e f i l e   l i n k k i t a p p @ m k 3 2 3 9" in "C:\Users\qijiahai\Desktop\wulian\alios-things-AliOS-Things-rel_1.3.3\AliOS-Things"
---
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

这个是报错日志

Ellsion, 2019-06-21:
我也遇到了同样的问题，请问您解决了吗？

closed 2022-03-01 by skylarCai